retro16/acsi2stm #4: Read issues

Frederick321, 2021-02-16:
Have a nice day!

I built an ACSI2STM; on a soldered breadboard, in order to minimize the tracks lenght.

Following your advices from "Device detection fails. Boot fails as well." issue thread; I managed to enabled correct write operation.

But reading sometimes present problems; I detected this loading some hard disk enabled games; once the full coloured splash screen of a game presented several garbage pixel lines.

On a major testing, it appears that large files (greater than 1 MB) can be loaded with more succesful results than load many small files reaching the same amount of data.

It can be something related to disk access time? Or something related to the explanation about "If DRQ goes up before ACK goes down, pulseDrqSend is too fast..."

Machine is a STE with TOS 1.06, and "Good DMA chip"

retro16, 2021-02-23:
Hi, some people (including myself) have timing problems because of a change in behavior in the STM32 library. I am aware of these issues, probably related to the pulseDrq functions. This needs fine tuning, some people already help me finding the correct timing.
That timing issue is there because the STM32 is not fast enough to react to the ACK pulse so it has to do guesswork with a timer,  which introduces these problems if the delay is not perfect. Hopefully this will be solved quickly.
Can you list the name of the games with problems so I can try to reproduce the issue ?

Thank you for your feedback.

retro16, 2021-04-23:
Hi, If you still have issues, I recommend that you upgrade to the 2.0 beta I just uploaded. Beware that there are minor pinout differences.

closed 2021-04-23 by retro16