retro16/acsi2stm #5: High current draw

TzOk83, 2021-02-17:
I've encountered a problem after shorting unused CS pins to GND the MCU overheats. I've checked, and PA1..4 and PB0..1 seem to be configured as output and set high, so shorting them to GND results in excessive current draw (50mA per pin).

retro16, 2021-02-23:
Thanks for the feedback.
I reviewed my code and your result is a bit strange. Based on my code, these pins should stay in the INPUT_PULLUP state.
In the meantime, you can remove the pins you don't use from the sdCs array in the code and leave them open.

retro16, 2021-04-23:
Hi, If you still have issues, I recommend that you upgrade to the 2.0 beta I just uploaded. Beware that there are minor pinout differences. Now, unused CS pins are not grounded anymore, which should solve your issue.

closed 2021-04-23 by retro16