retro16/acsi2stm #6: Debug Printout?

Ulises74, 2021-03-25:
Hi @retro16,
First of all, GREAT JOB! UFF WAO, I am getting my feet wet and I am overwhelmed. 
Can you post a debug printout of a successful SD card install?
In my case ICD ID detect and ICD Install fail to recognize anything in the bus. I realized ICD was sending a VERIFY(6), to which I modified the code to respond with a command Success. The ACSI2STM was recognized as ADAPTEC 5500, but would fail to install: "Unable to determine drive size". In the debug, I realized a command Test Drive Ready is sent with 0x20 in the 2nd byte, to which the program errors out. I am about to add a correction to this, but I believe I might be going down a rabbit hole.
Can you post a successful DEBUG session.
What is it recognized as by ICD?
Thanks

Ulises74, 2021-04-02:
Hi, I finally made it work! I'd like to share my experience here in case it helps anyone else. Trying to keep it short.
Things I learned:
1- This works, no need to mess with the software, except the pulses.
2- Pick a good strategy for assembly.
3- Triple check connections.
4- Run various tests, have patience.
What I recommend:
1- Do not use lead free solder, just use normal lead solder - lower temp, smaller blobs, easier.
2- Do not use protoboard, go straight for pcb prototypying, too many connections can go loose.
3- 8 bit lines plus 4 control lines, 12 total. Each point of solder multiplies, I had 4 stages, 12x4=48, plus SD, etc, you get the point.
4- triple check connections between points. I checked connections, then cleaned the flux with a toothbrush, which moved a wire without me noticing= 1 week lost.
5- Use a good known power supply, add a capacitor to your board at power entry, 10 or more uF. Tantalum or SMD's are better than electrolytics, but the latter will work as well.
6- There is a typo on the STM Serial port to be used. PA9 is to be connected to the USB Serial bridge RX port (not PA11 as in the original instructions). PA10 is correct.
Tips:
Whether the ACSI2SD works or not depends on the timings for read and write in the DMA port. These are affected by the number of pulses in the pulseDrqSend and pulseDrqRead functions as well as the compile options.
Sequence- 
1-Pick a compile option (default, O1, O2, O3)
2- change the pulses in the pulseDrqSend. Start by going down; but you could go up as well. Stick with one.
3- check that device is "read". ICD IDcheck will check device names by performing a read in the DMA.  Name should be 'ACSI2STM SD 0 1938 MB   v1.1' for example. You can reprogram the device while IDCheck is waiting for input.
2- If not read, change pulse durations. If sometimes it reads, and sometimes it doesn't or shows garbage, you still need to change pulse duration, but probably by one or two pulses.
3- If Pulse durations don't make the device be read, then change the Optimization. and repeat from step 1.
4 - Once you have stable detection of IDCheck you can start with format ICDFMT.
5- Go over the Format, then Partition. This reads data from the disk and writes data to the disk, but it does not confirm that data was written correctly. At this point you can start the calibration process with the pulseDrqRead; same steps as before, except this time, you will use the formatting/partitioning sequence.
6- You will have calibrated the pulseDrqRead function correctly, if when ICDFMT exits, it reboots the computer. ICDFMT during the partitioning process, will make the drive bootable and install a driver. So when it exits, it will try to read this back and reboot. If it doesn't reboot, it didn't work. If it reboots, you might just be near the sweet spot. If the ST booted to your C drive on screen, try creating folders and copying files, try reading/executing them. If you encounter errors, keep tweaking the pulses until you no errors are seen.
7- At this point you should have a working drive.

Notes:
- The watchdog timer lets you know something is wrong. If in the debug screen you see the welcome screen popping up a lot, means you have problems. At first, I had a lot of them going on. They went away when I added the power supply capacitor. Then I had some left during read/writes because I had changed the read/write subroutines. It meant, the sequencing wasn't done correctly and the STM was left waiting and after a while the watchdog timer rebooted the STM.
- I also noticed that seems like the DMA chip can tell if the hardware handshake is way out/not done in sequence, only for DMA writes to the drive. It will display "write error" during the partition creation or if you do a surface scan.
- Leaving the Debug on was crucial for me to learn/understand but you may not needed it if the above steps work. If you can tell that commands are received, you know that the soldering worked. Also, having the debug on did not interfere with the drive working, it just increased the access time by a bit. Remember to turn it off when done.
- Once the drive is working, you might be tempted to keep reducing the pulses. Each pulse is about 30ms. In my case, there was a point in which reducing the pulses just created errors. I believe you should do this in order to find the lower limit, and then leave the number of pulses in between the lower and the higher limit. I suspect temperature might affect DMA pulsing. 
- Compiling option that worked for me: Fast-O1
- My final numbers: 363KB/s, 2ms. Faster than an original Atari drive, not bad! (on ICDpro RateHd)
- I used a "quality" blue pill ($7), like the one in Retro16 pics. I will try on cheap blue pills ($2).

ToDo:
- I like it, I think I'll make a PCB, I have 2 Atari's, and it would suck to proto build these.
- I suspect the STM Roger Clark library is creating bloated assembly - SD runs at 25MHz so we should get faster speeds. In fact, I did away with the while(!ACK) wait in the pulse functions and it doesn't change the speed or the functioning. So I might dive into that - add inline assembly for port writing and reading.
- Enjoy it. I need to find ways to use the drive for gaming or programming or whatever.

Thanks Retro16!

Some Debug printouts:

-----------------------
ACSI2STM SD bridge v1.1
-----------------------

Initializing ACSI bus ...
ACSI bus ready
Initializing SD card 0
Size: 1938MB - 3970048 blocks
1 SD cards found

--- Ready to go ---
[... cut ...]
Size: 1938MB - 3970048 blocks
SD 0 ID ='ACSI2STM SD 0 1938 MB   v1.1'
Send: 0 0 1 0 1F 0 0 0 41 43 53 49 32 53 54 4D 20 53 44 20 30 20 31 39 33 38 20 4D 42 20 20 20 76 31 2E 31
Success
ID Shown!

ICDFMT

Command  12 0 0 0 24 0
Initializing SD card 0
Size: 1938MB - 3970048 blocks
SD 0 ID ='ACSI2STM SD 0 1938 MB   v1.1'
Send: 0 0 1 0 1F 0 0 0 41 43 53 49 32 53 54 4D 20 53 44 20 30 20 31 39 33 38 20 4D 42 20 20 20 76 31 2E 31
Success
Command  0 0 0 0 0 0
Initializing SD card 0
Size: 1938MB - 3970048 blocks
Success

Detected!

retro16, 2021-04-23:
Hi, If you still have issues, I recommend that you upgrade to the 2.0 beta I just uploaded. Beware that there are minor pinout differences. It should solve any timing issues as it now uses STM32 timers and DMA for much safer operation.
Thank you for the very detailed feedback that helped me a lot finalizing this v2.0.

closed 2021-04-23 by retro16