retro16/acsi2stm #8: Similar project, share code around image file usage?

mzryz, 2021-05-08:
Hi there,

Sorry if this is a silly question! But there is a SCSI>SD project called BlueSCSI also based on the STM32. Due to the similarities I was thinking this project may benefit from some of their code? For instance, they also support image files but they don't advertise any kind of slowdown in using them. I wonder where the difference is. I guess with the beauty of open source you guys might be interested?

https://github.com/erichelgeson/BlueSCSI

Thanks,

retro16, 2021-05-08:
Hi, this is very good information, thanks for the pointer.
I quickly looked at their code and they only use images and the exact same library, not raw SD card access, the performance they have is somewhat similar to my project when I use images. I bet that they would actually benefit from raw SD card access.

mzryz, 2021-05-08:
Ohhh okay, in that case I might let them know this. (I am a user of both asci2stm & bluescsi) I was just very interested in how both these projects are quite similar.

I will be modifying my asci2stm to v2 tonight and trying the new firmware ;)

closed 2021-05-08 by mzryz

erichelgeson, 2021-05-14:
Sorry to come to this late - but i see you're using shared SPI - have you tried dedicated? quite the speed bump if your project can support it.