retro16/acsi2stm #9: Advice on re-working to v2?

mzryz, 2021-05-09:
Hi there,

I have a couple of ASCI2STM's that @mst kindly made for me a while back, recently there has been an update to the device that requires some re-working. I haven't done much with SD cards and STM's so the instructions confuse me a little so I'm hoping to get some advice.

 So I can break the traces and patch these quite easily:
[image]
 Then the SD card info confuses me a lot, could someone help draw on my picture some of the changes I might need?
[image]

retro16, 2021-05-09:
Hi,
I confirm that the changes on the first picture are correct.
For the SD card, many STM32 pins had to be shorted to GND and now they must be disconnected.

* No change need to be made on the MicroSD adapter
* Since you have only one SD card, pins PA3, PA2, PA1, PB0, PB1 and PB3 of the STM32 are probably connected to GND (I can't see, the traces are either under the STM32 or under the black goo blob). The simplest way is to cut the pins directly on the STM32 itself as they are now completely unused.

mzryz, 2021-09-29:
Just closing the thread as I buggered it all up and my boards went in the bin.

closed 2021-09-29 by mzryz