retro16/acsi2stm #10: Debug serial port not correctly assigned

salacpavel, 2021-06-08:
When debugging my ACSI2STM hardware, I have realized there is no output on serial port (pins PA9, PA10). Even with 
#define ACSI_DEBUG 1   ...and...
#define ACSI_VERBOSE 1
the monitor is completely silent. Then I have realized the port is (incorrectly) identified as "Serial", while it should be Serial1.
Once I have replaced all instances of Serial with Serial1, I now see output messages in monitor, as expected.

midicdj1000, 2021-11-30:
HI, I have the same problem it sends nothing to the serial terminal and I tried changing Serial to Serial1.
If I use another program like the sdfat card info example it works perfectly on Serial.

retro16, 2021-12-08:
Hi, I have noted that this happens when ST-Link is used to program the device. I use the serial port for programming myself to work around this issue.

midicdj1000, 2021-12-10:
This was a big help, the diagnostic are showing correctly over the fdti programmer.

closed 2023-06-04 by retro16