retro16/acsi2stm #13: Wiring Diagram or Schematic?

VonHirsch, 2021-10-04:
Hello, is there a wiring diagram or schematic available for the build?  If not that's fine, I think I can figure out what goes where based on the descriptions, just wondering as it would be helpful.  Thanks for making this possible.

retro16, 2021-10-05:
Hello, there is currently no schematic for the build. Building the unit is essentially connecting pins described in the pinout tables of the "Building the ACSI cable" and "Connecting the SD cards" sections of the README and providing 5V to the STM32.

I believe that a schematic would look like a mess of confusing wires but if anyone feels otherwise and posts a schematic here I'll be happy to reference it in the official documentation with proper credits.

mlorenzati, 2021-11-30:
I can create a with the schematic, the PCB has I made has no sense since is for prototype pcb
[image]

retro16, 2022-06-28:
Version 3.00 now comes with an official schematic, PCB and Gerber. Thanks @mlorenzati as your schematic style helped me choose the clearest style for a schematic (mostly labels, just like what you did here). Closing this issue.

closed 2022-06-28 by retro16