retro16/acsi2stm #14: External LED PC13?

mamejay, 2021-11-03:
I have made up a board and connected a LED to PC13.
It is working but I find that it is on all the time and then flashes during disk activity.
I am using a regular blue pill board.  Looks like its inverted with the activity light on the Blue Pill.  I suspect its something simple.
Thanks

retro16, 2021-11-03:
Hi,
The internal LED of the blue pill is inverted, so you will have to do the same.
To connect your external LED, connect the anode to VCC and the cathode to PC13. Don't forget to add a resistor in series (at least 1k because the STM32 can't output much current).

mamejay, 2021-11-03:
Thanks for that.  I worked it out.  Unfortunately I made up a few PCB's so no clean way to fix this.  Anyway lesson learnt - always prototype your boards.

retro16, 2021-11-03:
You can still pretend it's a combined power+activity LED to turn it into a feature ;-)
Wish you the best with your project
I'm closing the issue if you don't mind.

closed 2021-11-03 by retro16