retro16/acsi2stm #15: CH32F103 - Anyone managed to get this working on this board?

mamejay, 2021-11-12:
I was sent CH32F103 boards instead of blue pills.
They program up the same but I cannot get them to work.  My ST just boots straight to the OS very fast but does not load from my SDCard.
Has anyone managed to get these working?

juancrix, 2021-11-15:
I received some boards with the same CH32F103. To program them I used their app (WCHISPTool) and the lower USB plug of the board. At least they worked with the old firmware with my v1 custom pcb. 
I haven't been able to test the CH32F103 with the lastest firmware and my v2 custom pcb, because I have not been abble to get an error free compilation, probably due to problems with my PC; if you don´t mind to share your compilation, I can test it with my configuration.

mamejay, 2021-11-15:
[acsi2stm.ino.generic_stm32f103c_WORKING.bin.zip](https://github.com/retro16/acsi2stm/files/7535527/acsi2stm.ino.generic_stm32f103c_WORKING.bin.zip)
Thanks for the comment.  Attached is my working compile that I made that works on original STM32F103C boards.  Let me know how you go.  Thanks!

juancrix, 2021-11-16:
My result with the CH32F103: it goes straight (and very fast too) to the desktop showing no icons (neither drive A nor drive B, nor trash icon). Just to confirm, can you tell me for how many sd cards is your compilation? and if there is any changes to the base firmware?

mamejay, 2021-11-16:
@juancrix I only use a single SDcard which I created from an image I got from patari.
I get the exact same results where is loads into the desktop fast.
As for the firmware no changes have been made at all.  Just compiled and uploaded to the CH32F103

luciust, 2022-04-03:
@mamejay Which version is your compile based on? It seems to work on my STM32 (supposedly an original) but it mostly discovers as 7xADAPTEC... can you share your compiled binary of 2.40? The released/compiled versions do not seem to even start on my STM32 chip :(

mamejay, 2022-04-03:
> @mamejay Which version is your compile based on? It seems to work on my STM32 (supposedly an original) but it mostly discovers as 7xADAPTEC... can you share your compiled binary of 2.40? The released/compiled versions do not seem to even start on my STM32 chip :(

@luciust I only have the older compiled bin file 2.1 file.  I need to try 2.4 out and see if I can get these many many clones I have here collecting dust.  Here is 2.1 which works on my real Bluepills.
[acsi2stm.ino.generic_stm32f103c_WORKING.bin.zip](https://github.com/retro16/acsi2stm/files/8405631/acsi2stm.ino.generic_stm32f103c_WORKING.bin.zip)

luciust, 2022-04-03:
@mamejay - Yup this is the one which shows up as 5 devices and ADAPTEC with mine. But at least it starts on mine. Well need to desolder two other bluepils and try again on fresh PCB...

retro16, 2022-04-04:
Hi, if any of you can manage to make 2.40 at least boot on CH32 or other clones, I would be very interested in your setup details to investigate why the release binary doesn't work on these.
Also, I wanted to suggest acsitest.tos (found in the release zip package, *tos* subfolder) that can help diagnosing 2.40 boards, in combination with your usual tools. In ACSI_DEBUG/ACSI_VERBOSE modes, the STM32 even outputs faulty bytes for DMA transfers. I might recognize some patterns I saw when doing DMA improvements/fixes for 2.3/2.4.

luciust, 2022-04-04:
@retro16 
I'm pretty sure I got a clone, since my second blue pill is from better source and has correct markings. I'm about to solder the gerber that has been forked from your repo - I hope the pin changes on 2.40 are not making this a problem.

Also dealing with issue that only -O2 option allows the 2.4 to compile, any other setting results with error. Latest STM32, Arduino 1.8.15 - on Linux... But that's another topic :)

mamejay, 2022-04-04:
@retro16 I tried many times to use the new code with a CH32 board but still get the desktop loading instantly.  WIth the ACSITEST.TOS is this a TOS replacement that needs to be burnt to an eprom?

@luciust here is the 2.40 bin file that I managed to compile and get working on my "real" bluepill.  See if this is better than the last binary I sent
[acsi2stm.ino.generic_stm32f103c_2.40_WORKING.bin.zip](https://github.com/retro16/acsi2stm/files/8409128/acsi2stm.ino.generic_stm32f103c_2.40_WORKING.bin.zip)

retro16, 2022-04-04:
@mamejay ACSITEST.TOS is an Atari ST program. Just run it from the GEM desktop. You can load it by any method (working acsi2stm, floppy, ultrasatan, ...) it will work the same.
The program supports hot plugging fine, so you can load it using one ACSI device, unplug it and plug the device to test. You can test multiple devices in a row or reflash your acsi2stm firmware and try again by leaving the program running on the ST.
I don't guarantee that you can quit back to GEM cleanly if hot swapping, but who cares ...

luciust, 2022-04-04:
@mamejay Great, will test it with old one and hopefully finish solderiing new one.

luciust, 2022-04-04:
@mamejay - So, no joy, the 2,40 from you also doesn't start on the clone. No green LED, no activity. Also I I compile debug version with serial - not even serial activity. 

Soldering the other board then!

retro16, 2023-06-04:
If anyone has success with 4.00 on a CH32 or any other clone, please drop a line !
Thank you for your help, you people really rock.

mamejay, 2023-06-04:
> If anyone has success with 4.00 on a CH32 or any other clone, please drop a line ! Thank you for your help, you people really rock.

I have heaps of them still.  They need to be programmed a strange way so I will see if I can actually get the  code onto one and use my ZIF socket test rig

mamejay, 2023-06-04:
Tried acsi2stm-4.00.ino.bin and also acsi2stm-4.00-strict.ino.bin get the following as soon as I switch on the ST.
Information just flashes vertically down the screen.
If there is anything else you want me to try let me know.
Thanks

[image]
[image]

retro16, 2023-06-12:
Okay, this doesn't look very good. Let's say that these clones are officially not supported.
Leaving the issue opened for reference as it contains useful information. If I ever update the documentation, I may integrate that thread in the troubleshooting section and close this issue.