retro16/acsi2stm #16: SPI Bus is too fast

mlorenzati, 2021-12-06:
After checking the proyect I saw random behaviors in the SPI bus, like failing to start the **SdSpiCard** or detecting 0MB of free space.

After toying with the solution, I tested to lower the SPI to 18Mhz and delay a bit prior init.

```
if(!card.begin(SdSpiConfig(csPin, SHARED_SPI, SD_SCK_MHZ(18), &SPI))) {
    resetState();
    return false;
  }
  delay(250);
  blocks = card.cardSize();```

Now it always boot the sd and detects a boot image on it.
Now dealing with a bus collision that stops the stm32

retro16, 2021-12-07:
Thanks a lot for the feedback. I had strange issues too and got around by doing retries, which was obviously the wrong way.
I will set 18MHz by default and add a #define for adventurous people who want 25MHz back.

I'm rather busy on other projects lately so it might take a while to do that as my Atari is not set up anymore.

retro16, 2022-02-27:
Thanks a lot @grzybsonssg for the feedback. If that works for you too and it's stable, please give some feedback and I'll just commit @mlorenzati 's solution and do a new incremental release.

retro16, 2022-03-01:
The patch here looks good so I integrated it in my latest release with a #define for people who want the highest speed possible. Thank you all for the feedback.

closed 2022-03-01 by retro16