retro16/acsi2stm #17: Bus collision on ST power up

mlorenzati, 2021-12-07:
I was able to separately test the board, it reads properly the SD card as a boot device and waits for ACSI bus.
But If I connect the device to the db19, if I power the ST, the green led goes down and the USB serial connection is lost with the pc.

Which lines might be colliding?
this is my schematic, which honors the same documentation

[image]

retro16, 2021-12-07:
Hi Marcelo,

I just had a very quick look at your schematic, it looks very good overall (I trust your pin assignments, I'm sorry I don't have time to double check connections right now, and I'm pretty sure that you double-checked these already).
The only issue I can think of is the USB data lines on PA11/PA12. If you power the STM32 with its onboard USB plug, make sure to use a cable that is not connected to the data lines.

Having the LED turning off after detecting the ACSI bus is good sign, the LED turns on while waiting for the bus, then turns into an activity LED when entering the main loop.

Of course losing the serial connection is not good at all. What happens if you reset the STM32 with the ST already powered on ?

mlorenzati, 2021-12-07:
Oh so you mean that the USB connection is only meant for powering the device?
What about the COM port? When I debug is using the serial through USB, which serial should be configured?

retro16, 2021-12-08:
I'm surprised that you even have a COM port to begin with. This project is designed to work with USART1 (PA9+PA10) connected to an external UART-USB converter. Can you check if you have a compilation setting that sets the behavior of Serial (I don't remember it that well and my Arduino setup is not reachable right now): it should be set to use hardware serial and mapped to USART1.
It is not possible to use the internal USB port at all because the code requires the TIM1_ETR pin for its very specific behavior that cannot be remapped anywhere else.

mlorenzati, 2021-12-08:
I'll check why is setting the USB2Serial port, but I can assure you that it was wrongly configuring that and was able to see the Serial output through there.
I think is related to use the maple mini interface

mlorenzati, 2021-12-08:
Now that I program it through serial I can see the logs that detects the disc as a boot device, and acsi says ready, but the machine does not starts the disk
```
-----------------------
ACSI2STM SD bridge v2.1
-----------------------

SD0:ACSI2STM SD0   30 GB F BOOT
1 SD cards found
Waiting for the ACSI bus

--- Ready to go ---

```

mlorenzati, 2021-12-08:
This is the DMA dump of sector 0 with commands 8(read) and 0(unit ready)
http://fr.meric.free.fr/Technique/Hardware/memoires%20de%20masse/ACSI_DMA_Guide_6-28-1991.pdf
It appears to be ok, but no clue why it does not continue
```
[+8]SD0 cmd 8 [<0][<0][<0][<1][<0](6 bytes)
Read 1 blocks from 0
DMA send (512 bytes): 60 1E 0 0 0 0 0 0 50 54 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 48 79 0 B FF FF 4E 4D 58 8F 8 0 0 3 67 24 42 78 4 46 2F 3C 0 E 0 0 4E 41 58 8F 31 FC 0 88 86 6 21 FC 0 0 0 88 86 4 31 FC 0 8A 86 6 4E 75 20 78 4 76 C A8 50 50 48 44 FF F8 67 CE 2F 7 48 79 0 0 BB 80 3F 3C 0 48 4E 41 5C 8F 20 40 D1 FC 0 0 80 E4 72 2 74 A 7C 0 4D F8 86 6 4B F8 86 4 50 F8 4 3E 48 50 11 EF 0 3 86 D 11 EF 0 2 86 B 11 EF 0 1 86 9 58 8F 3C BC 1 98 3C BC 0 98 3A 82 3C BC 0 88 30 3C 0 8 80 6 48 40 30 3C 0 88 61 0 0 72 66 0 0 8A 20 1 30 3C 0 8A 61 0 0 64 66 0 0 7C 20 1 E0 48 48 40 30 3C 0 8A 61 0 0 52 66 0 0 6A 20 1 48 40 30 3C 0 8A 61 0 0 42 66 0 0 5A 20 2 48 40 30 3C 0 8A 61 0 0 32 66 4A 2A BC 0 0 0 A 38 3C 1 90 61 0 0 26 66 3A 3C BC 0 8A 30 15 2 40 0 FF 66 0 0 2E 3C BC 0 80 51 F8 4 3E 3F 6 4E 90 3C 1F 60 32 2A 80 78 A D8 78 4 BC 8 39 0 5 FF FF FA 1 67 8 B8 78 4 BC 66 F0 78 FF 4E 75 3C BC 0 80 BC 3C 0 E0 67 8 6 6 0 20 60 0 FF 24 51 F8 4 3E 2E 1F 4E 75 2F 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 C4 F3 63 FA B6 65 0 1 1 0 6 93 36 1D 3F 0 0 0 0 40 7 0 0 94 36 1D F 40 66 CB 7E 40 7 0 7A 41 34 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 55 AA OK
Success
[>0][+0]SD0 cmd 0 [<0][<0][<0][<1][<0](6 bytes)
Success
[>0][+0]SD0 cmd 0 [<0][<0][<0][<1][<0](6 bytes)
Success
[>0]
```

mlorenzati, 2021-12-15:
Does the sequence looks ok? I heard that the issues with the current 2.0 approach are related to TIMER1/TIMER4 on the available STM32F103C8 on the bluepills on the market, which produces ACSI reading errors on the STM.

retro16, 2021-12-15:
Hi, sorry for the long delay.
This log looks okay. It's strange that you have 2 cmd 0 after that, though it might be because of an error.
It's likely that there is some data corruption somewhere in DMA transfers, there are a lot of possible causes (that can be ST issues, STM32 hardware issues, a bug in the STM32 code, connection issues, ...). Without signal traces it's difficult to conclude.

Can you try changing ACSI_ACK_FILTER in acsi2stm.h between 0, 1 and 2 ?

mlorenzati, 2021-12-15:
I will and return to you asap :)

mlorenzati, 2021-12-15:
with ACSI_ACK_FILTER in 0, I just see
``` [+0]SD0 cmd 0 [<0][<0][<0][<1][<0](6 bytes)
Success
[>0]```

with ACSI_ACK_FILTER in 1, I see
```[+8]SD0 cmd 8 [<0][<0][<0][<1][<0](6 bytes)
Read 1 blocks from 0
DMA send (512 bytes) OK
Success
[>0]```

with ACSI_ACK_FILTER in  2 I see
``` [+8]SD0 cmd 8 [<0][<0][<0][<1][<0](6 bytes)
Read 1 blocks from 0
DMA send (512 bytes) OK
Success
[>0][+0]SD0 cmd 0 [<0][<0][<0][<1][<0](6 bytes)
Success
[>0]```

But it's also dependent of SPI speed for SD, best results have been seen with ACSI_ACK_FILTER in 1

mlorenzati, 2021-12-15:
[image]
I tested with a AHPT94, and it properly transfers (280-310KB/s) with 9.8mS access time
But I cannot boot with the device, can it be the TOS version? I have 1.02

mlorenzati, 2021-12-15:
This are the logs with AHPT94, I think they look ok, no error is reported in the app, but the level of test is just Direct HW access, could it be the TOS of the ST520?
``` 
[+0]SD0 cmd 0 [<0][<0][<0][<0][<0](6 bytes)
Success
[>0][+0]SD0 cmd 0 [<0][<0][<0][<38][<0](6 bytes)
Success
[>0][+8]SD0 cmd 8 [<1F][<FF][<FF][<1][<0](6 bytes)
Read 1 blocks from 2097151
DMA send (512 bytes) OK
Success
[>0][+8]SD0 cmd 8 [<0][<13][<88][<1][<0](6 bytes)
Read 1 blocks from 5000
DMA send (512 bytes) OK
Success
[>0][+0]SD0 cmd 0 [<8][<13][<7B][<1][<0](6 bytes)
Success
[>0][+8]SD0 cmd 8 [<0][<16][<C3][<1][<0](6 bytes)
Read 1 blocks from 5827
DMA send (512 bytes) OK
Success
[>0][+8]SD0 cmd 8 [<8][<16][<B6][<1][<0](6 bytes)
Read 1 blocks from 530102
DMA send (512 bytes) OK
Success
[>0][+0]SD0 cmd 0 [<0][<19][<FE][<1][<0](6 bytes)
Success
[>0][+8]SD0 cmd 8 [<8][<19][<F1][<1][<0](6 bytes)
Read 1 blocks from 530929
DMA send (512 bytes) OK
Success
[>0][+0]SD0 cmd 0 [<0][<1D][<39][<1][<0](6 bytes)
Success
[>0][+0]SD0 cmd 0 [<8][<1D][<2C][<1][<0](6 bytes)
Success
[>0][+8]SD0 cmd 8 [<0][<20][<74][<1][<0](6 bytes)
Read 1 blocks from 8308
DMA send (512 bytes) OK
Success
[>0][+0]SD0 cmd 0 [<8][<20][<67][<1][<0](6 bytes)
Success
[>0][+0]SD0 cmd 0 [<0][<23][<AF][<1][<0](6 bytes)
Success
[>0][+8]SD0 cmd 8 [<8][<23][<A2][<1][<0](6 bytes)
Read 1 blocks from 533410
DMA send (512 bytes) OK
Success
[>0][+8]SD0 cmd 8 [<0][<26][<EA][<1][<0](6 bytes)
Read 1 blocks from 9962
DMA send (512 bytes) OK
Success
[>0][+0]SD0 cmd 0 [<8][<26][<DD][<1][<0](6 bytes)
Success
[>0][+8]SD0 cmd 8 [<0][<2A][<25][<1][<0](6 bytes)
Read 1 blocks from 10789
DMA send (512 bytes) OK
Success
[>0][+8]SD0 cmd 8 [<8][<2A][<18][<1][<0](6 bytes)
Read 1 blocks from 535064
DMA send (512 bytes) OK
Success
[>0][+8]SD0 cmd 8 [<0][<2D][<60][<1][<0](6 bytes)
Read 1 blocks from 11616
DMA send (512 bytes) OK
Success
[>0][+8]SD0 cmd 8 [<8][<2D][<53][<1][<0](6 bytes)
Read 1 blocks from 535891
DMA send (512 bytes) OK
Success
[>0][+8]SD0 cmd 8 [<0][<30][<9B][<1][<0](6 bytes)
Read 1 blocks from 12443
[... 1479 lines cut ...]
DMA send (512 bytes) OK
DMA send (512 bytes) OK
... repeating same log
DMA send (512 bytes) OK
DMA send (512 bytes) OK
DMA send (512 bytes) OK
DMA send (512 bytes) OK
DMA send (512 bytes) OK
Success
[>0][+8]SD0 cmd 8 [<0][<51][<18][<A0][<0](6 bytes)
Read 160 blocks from 20760
DMA send (512 bytes) OK
DMA send (512 bytes) OK
DMA send (512 bytes) OK
DMA send (512 bytes) OK
... repeating same log
DMA send (512 bytes) OK
DMA send (512 bytes) OK
DMA send (512 bytes) OK
DMA send (512 bytes) OK
Success
[>0][+0]SD0 cmd 0 [<0][<51][<B8][<A0][<0](6 bytes)
Success
[>0]
```

retro16, 2021-12-16:
Does AHPT94 check data integrity ?
Commands and data are transmitted using very different methods so you could have perfectly valid command bytes and big errors in data transmission.

It would be interesting to use a tool that displays the raw content of the device. I don't know anything like that so I don't have one to recommend, unfortunately.

I don't think TOS version has any impact on the boot process. It's basically downloading the block using the DMA chip (so TOS does nothing during that phase), doing a checksum, and running the block as-is if the checksum matches 0x1234. Your SD card boot block has the correct checksum because it is listed as bootable on the debug output.
TOS only has an influence on the filesystem implementation (FAT/FAT32/...).

mlorenzati, 2021-12-16:
Sadly I could test that the app does not validate integrity, and validated that the data is not right (e.g. the disk size varies from right to wrong). So reading commands might be right (st to stm) but wrong the sent data (stm to st).

retro16, 2023-06-04:
This looks like the infamous STM32 hardware bug that plagued the project for years ! If 4.00 doesn't fix, please reopen.

closed 2023-06-04 by retro16