retro16/acsi2stm #18: Migrate to a Raspberry pico

mlorenzati, 2021-12-15:
I've seen the documentation that states that the main problem is related to timing when checking the CS pin.
Rasberry pico can run from 130mhz to up to 300mhz overclocked and in the same form factor of the bluepill, plus the rpi has a really powerful programmable PIO that works with DMA, independently from the M0 cores if required.

How feasible would be to transition to that soc?

retro16, 2021-12-15:
It is doable. Basically you would have to rewrite the Acsi class in Acsi.h to adapt it to the Pico.

I'm not using this project actively anymore so I don't really have time to involve in such a project beyond answering a few questions, but I could probably help somebody who wants to do this by giving a few hints. Implementing the ACSI/DMA protocol on the Pico PIO is a good way to achieve higher performance and reliability than this STM32 implementation.

To be honest the TIMER approach in v2.1 should really work 100% on paper and yet some people still have issues. Making this really work reliably would require a lot of testing on a lot of different ST from many generations.

retro16, 2023-06-04:
Closing that old issue as 4.00 fixes the issue mentioned.

closed 2023-06-04 by retro16