retro16/acsi2stm #19: Gerber files

jeroentaverne, 2022-01-17:
Hi. Are there some gerber files available for this project? I know it can be hand wired, but a real PCB will make it look much nicer. Thanks!

tomtebloss, 2022-02-01:
I did fork this out and made some PCB: https://github.com/tomtebloss/acsi2stm/tree/master/gerbers

luciust, 2022-04-03:
Hey, thanks for the gerbers, just a note: the R3 is 10k not 100k. With 100k you can't set the programming mode. BOM needs fixing pictures are showing correct resistor.

tomtebloss, 2022-04-03:
Thx for pointing it out. Changed BOM and readme. I changed R4 to 10K as well since it is also part of the programming mode but not really used in out project.

luciust, 2022-04-03:
@tomtebloss No problemo, big thanks for gerbers - I've finished soldering first one and now trying to cope with my STM32 version which is odd - seems to be either not starting the official build or my compiled version, but kind-of works (shows up as multiple ADAPTEC 5500 controlers) with this binary:  https://github.com/retro16/acsi2stm/issues/15#issuecomment-968469499

luciust, 2022-04-05:
@tomtebloss One more question - the R2 on your photo is 1K while BOM states 10k. Should it be 1k as well?

tomtebloss, 2022-04-05:
> @tomtebloss One more question - the R2 on your photo is 1K while BOM states 10k. Should it be 1k as well?

I'm unsure which photo you mean? Schematics says 10K and the [photo](https://github.com/tomtebloss/acsi2stm/blob/master/gerbers/images/ACSI2STM_v1.2_back.jpg) is 1002? A 10K right?

Anyway, it's part of the reset circuit along with C2.

luciust, 2022-04-05:
@tomtebloss Yeah - correct the 4 digit mode - yet if photo is correct - then BOM says: R2, R3 are 10k , while on the said picture we have R2 @ 10k and R4 @ 100k. My fault I got confused with my 3 and your 4 digit code - but I would say that BOM and R2,R3 and R4 vs picture - need your eyes. I'm debugging the fake STM32 - thus looking at each small piece on blue pill vs gerber too...

tomtebloss, 2022-04-06:
Here is the [schematics](https://stm32-base.org/assets/pdf/boards/original-schematic-STM32F103C8T6-Blue_Pill.pdf) I used (if I remember correctly) of the "bluepill" when I designed my board. Apparently R10 should be changed to 1.5K. So I will probably change that at least. I didn't use USB for programming. Only for power.

I will have to change some resistors and try mine out first and then take new photos.

luciust, 2022-04-06:
Thanks! I needed that!

retro16, 2022-06-28:
Version 3.00 now provides an official PCB. Closing this issue.

closed 2022-06-28 by retro16