retro16/acsi2stm #20: Problems building for STM32F401CCu6 clone BlackPill

z80micro-mc, 2022-02-28:
I am having problems trying to build for a Blackpill clone - WeAct v3 type - with 3 oval buttons, STM32F401ccu6 processor, LED on PC13.  I am running Arduino 1.8.12 with STM 2.2 boards

I can build and upload simple sketches ok. (eg Blink ) But keep getting compile errors:

mapledma.h

Acsi.cpp:240:10: fatal error: libmaple/dma.h: No such file or directory
240 | #include <libmaple/dma.h>
| ^~~~~~~~~~~~~~~~
compilation terminated.

exit status 1
libmaple/dma.h: No such file or directory

Not sure why it wants the maple library it is a Blackpill not a Bluepill.

I have the SDfat library installed and have created a Hardware directory with the STM32 files in it.

[image]
[image]
[image]
[image]

[image]

Any input would be appreciated.

Thanks

Peter

retro16, 2022-03-01:
Hi, this projects doesn't use the official STM32 board (selected in your last screenshot).
It uses the Roger Clark Melbourne framework (for historical reasons) available here: https://github.com/rogerclarkmelbourne/Arduino_STM32
This framework is based on the Maple framework, that's why you are missing libmaple includes.

Once it compiles, you will probably need to reassign pins based on the requirements of the integrated timers. Also, I cannot certify that the DMA input trick using an unused timer register will work on anything other than the original STM32F103C8. It does pretty wild stuff with timers and DMA to achieve stable and fast results.

JM

z80micro-mc, 2022-03-01:
Jean,

Thanks for the reply. I take it from this that the 'BlackPills' that you 
refer to are the older STM32F103 based variants and not the newer 
STM32F401 / STM32F411 based versions.

I had hoped that as the notes suggested that the main issue was with LED 
on PB12 instead of PC13 being the most serious issue with the BlackPill 
so when I found the newer STM32F401 based versions with the LED already 
on PC13 I was hoping for a simple fast solution this together with the 
greater availability of the newer boards in the UK.

Peter

clone BlackPill (Issue #20)

or Android

retro16, 2022-03-01:
Oh you are absolutely right, I was refering to the older F103 black boards (which seem to be very rare). This needs an adjustment of the README to make it clear. Thanks a lot for bringing these new boards to my attention (I'm working on non-microcontroller projects these days and didn't keep up with the newer stuff).

retro16, 2022-03-01:
Updated the README. Keeping the issue opened if anyone want to try to make this project run on the newer black pill boards.