retro16/acsi2stm #21: Low memory resources warning when building for BluePill

z80micro-mc, 2022-03-06:
I have just successfully compiled the ACSI2STM code for a BluePill STM32F103C8T6. However, I get compiler warnings that the resulting code may be unstable due to low availability of memory resources.

It informs me that 83% of Flash is used 54882 bytes and 77% of RAM 15880 bytes. (-O2 option)

However, ST-Link informs me that the STM has 128 kb Flash so less than 1/2 full.

[image]

As I understand it an STM32f103C8 has 64 kb flash but ST-link is reporting that mine has 128 kb.

Is this a problem?
Do I need to upgrade to an STMF103CB (128 kb flash). Not sure how else to resolve this as my attempts to use STMF401CC BlackPill fails to compile.

Thoughts

Thanks

Peter

z80micro-mc, 2022-03-06:
As ST-Link is informing me that I have 128 k flash, I have just tried building as an STM32F103CB

It is still not happy with low memory though now only 41% of Flash is in use, still 77% or RAM.

**Adruino IDE o/p**
_Sketch uses 54884 bytes (41%) of program storage space. Maximum is 131072 bytes.
Global variables use 15880 bytes (77%) of dynamic memory, leaving 4600 bytes for local variables. Maximum is 20480 bytes.
Low memory available, stability problems may occur.
STM32 ST-LINK CLI v3.3.0.0
STM32 ST-LINK Command Line Interface

ST-LINK SN: 56FF6F065080495143331687
ST-LINK Firmware version: V2J39S7 (Need Update)
Connected via SWD.
SWD Frequency = 4000K.
Target voltage = 3.3 V
Connection mode: Normal
Reset mode: Software reset
Device ID: 0x410 
Device flash Size: 128 Kbytes
Device family: STM32F10xx Medium-density

Loading file...
Flash Programming:
  File : C:\Users\Peter\AppData\Local\Temp\arduino_build_974230\acsi2stm.ino.bin
  Address : 0x08000000
Memory programming...
 0%������ 13%������� 27%������� 41%������� 55%������� 69%������� 83%������� 97%�� 100%
Memory programmed in 4s and 391ms.
Programming Complete._

So other than moving to STM32F4xx series not sure how else to get more RAM.

Peter

retro16, 2022-03-07:
Hi,

The warning message is because of low RAM.

About flash, there are 128k chips sold as 64k (I have a few too), but in any case there is no issue there because the code fits in 64k anyway. What you see is perfectly normal.

I did a few tests and it seems that the upload method changes the amount of avaiable RAM. Uploading via st-link eats up something like 600 bytes of RAM.

4600 bytes of RAM is still plenty (well, not really, but it's acceptable) to make this project run.

A few ways to decrease RAM usage:
* Comment out lines in the sdCs array for SD card slots you don't need.
* Remove the snprintf call that eats up a lot of RAM. That has been done in the "experimental" branch that I don't recommend trying that branch at all (it's basically work in progress but it came to a stall because I don't have time anymore to work on that project).

If RAM is really an issue, I might try to backport this snprintf removal thing but since I don't have access to a ST right now I'll need some beta testers.

retro16, 2022-06-28:
3.00 was optimized to use less RAM. Closing the issue.