retro16/acsi2stm #22: Not compatible with the standard, some drivers don't work

troed, 2022-03-09:
It seems the ACSI2STM project isn't implementing (or not correctly implementing) some commands that would allow it to work with more drivers on the Atari (in this case, HDDriver specifically). There's been a discussion over on the HDDriver forum and Uwe has helpfully suggested using his testsuite and monitor (see at the bottom of https://www.hddriver.net/en/scsidriver.html ) to figure out where the problems might be.

I've run SDRVTEST which complained on not getting an error accessing invalid LUN, and hanging on reading from the invalid LUN, as well as SERRTEST that threw an error on getting the capacity. I've not researched this further.

It is, sort of, possible to use HDDriver with ACSI2STM (see my post here: https://www.hddriver.net/forum/viewtopic.php?p=887#p887 ) so I don't expect it to be much work (famous last words) to increase compatibility.

retro16, 2022-03-14:
I just commited some changes into master. I didn't fully test hddriver, but at least SDRVTEST does not complain too much anymore. SERRTEST is still not happy.

retro16, 2022-06-28:
Version 3.00 should be much more SCSI compliant, especially in strict mode.

closed 2023-06-04 by retro16

fplanque, 2024-09-15:
> Version 3.00 should be much more SCSI compliant, especially in strict mode.

Does this mean it's "more" compliant but still not *really* compliant ?

I had many [many many issues](https://www.fplanque.com/dev/retro/atari-st-acsi2stm-mega-ste/) trying to get it to work in conjuction with an internal Mega STE SCSI controller.