retro16/acsi2stm #24: Too many messages on startup

TzOk83, 2022-03-21:
2.31 prints out a copyright/introduction message for every SD card, I think printing it once would be enough ;)

retro16, 2022-03-21:
2.31 only prints a message if no SD card is inserted or if the SD is not bootable.

You probably have less than 5 SD card readers, a way around that would be to set ACSI_SD_CARDS in acsi2stm/acsi2stm.h to the actual number of slots you have. If all card readers have a SD card inserted, you should see no message.

If the feature annoys you, just compile with ACSI_DUMMY_BOOT_SECTOR and ACSI_BOOT_OVERLAY set to 0 to turn everything off.

I'll think about a way to reduce this, but it's not trivial as each SD card is an independent device and the boot payload is the same for all devices.

TzOk83, 2022-03-21:
I have used a precompiled binary instead of compiling my own, so I had no chance to set anything.

Interesting fact - on my STfm it displays messages and doesn't autoboot, while on STe it displays only ICD splash, and autoboots. Same SD card, same everything.

retro16, 2022-03-22:
I am working on a new way to disable unused SD slots by soldering some pins, so people can build a hardware unit without having to rebuild from source on each upgrade.

closed 2022-03-26 by TzOk83