retro16/acsi2stm #26: Atari TT030 compatibility?

GreatStoneEx, 2022-06-15:
Hello,

My Atari TT030 is not able to boot on acsi2stm (which is fine, it was mainly for my 1040).
Is there any plan to make it compatible with TT?

Regards

retro16, 2022-06-15:
Hello, thank you for your feedback. I don't have access to a TT (in fact, I never saw a real TT !) so I cannot work on its compatibility. If it's only a boot issue, this might be doable on an emulator, does it work if you boot a driver such as ICD from a floppy drive ?

If anyone with good TT knowledge and a logic analyzer (and a lot of patience) is willing to cooperate on this task, I could work on this after releasing a stable 3.0.

If the TT hard drive port is any faster than the ST, then it's impossible to make it work because the ST already pushes that good old STM32 blue pill to its limits.

GreatStoneEx, 2022-06-15:
Thanks for the quick answer!
I have a gotek coming, to plug into the TT, so that I can easily test stuff like that: as soon as it is installed, I will try it to see if it's coming from the boot. Hopefully, it will be just that :)

GreatStoneEx, 2022-06-16:
So, I tested today with the new gotek.
* Image of ICD driver 6.55a in the selected image of the gotek
* acsi2stm plugged
If I turn on the TT, wait for some time, icd driver is booting and showing me a list of all the partitions of the sd card from the acsi2stm!!
Then arriving to the desktop, I am able to browse my files :)
Sim city is working like a charm!

When, I decided to shutdown the TT, remove the floppy image from the gotek, boot again.
...and it is also working now ?!

I do not know what was going one, but asci2stm is now working fine in my TT, directly when starting it :)

retro16, 2022-06-19:
Wow that is super good news. Thanks for the feedback. Boot can be a bit hit or miss depending on SD timings or other factors.
I will close this ticket, feel free to reopen if you encounter issues again on your TT.

closed 2022-06-19 by retro16