retro16/acsi2stm #27: Reliable source or Blue Pill boards

mamejay, 2022-07-16:
Not really an issue but I have had not luck getting working Blue Pills.  Does anyone have a trusted seller of Blue Pill boards that they can recommend?
Thanks

retro16, 2022-07-16:
I must apologize for a big mistake I just did. In my usual process of fighting spam, I wrongly reported this issue as my broken brain thought this was about another kind of "blue pills" !
Sorry mamejay for what I just did, I recognized your pseudo a bit too late. If there is any kind of aftermath, let me know if I need to do anything about it.

vitecd, 2022-07-17:
https://www.aliexpress.com/item/4001169143888.html
https://www.aliexpress.com/item/32377410710.html

mamejay, 2022-07-17:
Thanks for that Vitecd.  So have you purchased from this seller recently and they have been ok?

vitecd, 2022-07-18:
Yes, I have pills from both suppliers.
Cheers

mamejay, 2022-07-19:
OK.  I have ordered 5 of the second link you sent.  I find the super cheap ones are always duds.  Now we wait and see

mamejay, 2022-07-28:
Just to let everyone know.  Received the Blue Pills and they program fine except they will not boot.  I have so many of these duds now I am just giving up.

retro16, 2022-07-28:
I just order one from the second link of vitecd, hopefully I will receiv the same as mamejay. Once received (usually takes 1 month for shipping where I live) I will test it myself. If I have the same issues, I will try to debug this further, making bad units work would be very interesting.

mamejay, 2022-07-30:
Excellent.  I hope you can get these working.  I have only been testing with 2.x code as I have not modified any boards for 3.x code.
With the modifications for a 2.x board will 2.x firmware still work or do I need to just use 3.x firmware?  I just test my boards using a patari games image.  Will this still work on 3.x?

retro16, 2022-07-30:
You can safely flash 3.x to 2.x units. All you risk is having read-only drives and a little bit of unstability at reboot. Read-only problem can be worked around by patching ACSI_SD_WRITE_LOCK to 0 in acsi2stm.h.
3.x provides its own embedded driver so you don't need any image to test the unit, in fact you don't even need an SD card !

mamejay, 2022-07-30:
How do you know its working without a SDCARD?

retro16, 2022-07-30:
3.x embeds setup software and a boot driver directly into the STM32 firmware. You can access the setup tool by mashing the Shift+S keys during Atari boot. See the a2setup doc for more info. Of course, to load and save files or do anything useful, you need a SD card.

mamejay, 2022-07-31:
OK.  Did more testing of these boards.  I have always used a STMLink to program the boards and decided to use a serial USB.
My working board program fine with it but when I try with these new Blue Pills I get the follow.

C:\arduino-1.8.5_Portable\hardware\Arduino_STM32-master/tools/win/serial_upload.bat COM7 {upload.altID} {upload.usbID} C:\Users\Jason\AppData\Local\Temp\arduino_build_582505/acsi2stm.ino.bin 
Failed to init device.
stm32flash 0.4

http://stm32flash.googlecode.com/

Using Parser : Raw BINARY
Interface serial_w32: 115200 8E1

I suspect something is a miss with these boards.  Any advise?

mamejay, 2022-08-25:
> I just order one from the second link of vitecd, hopefully I will receive the same as mamejay. Once received (usually takes 1 month for shipping where I live) I will test it myself. If I have the same issues, I will try to debug this further, making bad units work would be very interesting.

Did you manage to get your hands on those boards yet?

retro16, 2022-08-28:
Hello, I received one this week from [Mega Semiconductor CO., Ltd.](https://fr.aliexpress.com/item/32377410710.html) and it is useless. Programs fine using serial, but the STE doesn't boot on it. In debug mode, it enters a strange soft reset loop that makes no sense. I didn't check more than that.
For programming issues, maybe reflashing the bootloader with Roger Clark's would restore serial programming: https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/binaries/generic_boot20_pc13.bin.

mamejay, 2022-08-29:
Far out!  Oh well at least it was not me.  I managed to buy 20 STM chips from LSCS for like $3.5USD but now they are out of stock.  I have just been replacing the chips on the shit boards with the genuine ones and they are working

retro16, 2022-08-29:
For the record, my bad STM32 chip has its text marking way off center and not 100% aligned, the font is also thinner than the other models. Not sure whether it's a sign of something smelly, but it confirms that there are some bad clones.
The PCB and other components were good quality though, it seems that the issue is in the STM32 chip itself.
I will keep this github issue opened as a sort of forum thread so people can give their feedback about their experience.

vitecd, 2022-08-29:
Magics around... Need to buy something fresh and fall to debugging

TzOk83, 2022-10-01:
I have recently bought STM32F103CBT6 chips from KINGDEVIL store, and they seem to work fine:
https://www.aliexpress.com/item/1005003586463686.html

Have you verified your BluePills with this:
https://mecrisp-stellaris-folkdoc.sourceforge.io/bluepill-diags-v1.640.html

Recently bought ST-Link V2 clones have some strange Apexmic chip, which rejects to work with SM32 Cube Programmer.

retro16, 2023-06-04:
Dropping a line to indicate that 4.00 works around a very sneaky (and semi-random) hardware bug on the STM32. Many units that didn't work reliably before will work correctly.
Also, to properly test a unit's stability, you can use `ACSITEST.TOS` with the buffer and command tests that spam DMA and command bytes.

It'll keep the issue opened as a reference as it may be an interesting topic for newcomers.

jfceklosky, 2023-07-18:
I have been using these from Amazon without any issues so far with 3.x and 4.x

https://www.amazon.com/gp/product/B07VKSVM21/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1

jfceklosky, 2023-08-20:
These are a another reliable source for asci2stm.  

I have personally tested the software with these, version 3.X and 4.X

https://www.amazon.com/gp/product/B0B3X4BH4Q/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1