retro16/acsi2stm #30: Wrong BPB

Dunkelwind, 2022-08-28:
Hello,
sysinfo.prg & SED (from Scheibenkleister) reports wrong BPB.
Here are the infos from SED:

```
Anzahl der Zylinder   : 2560
Anzahl der Kpfe      : 35
Breite der Parkzone   : 0
Steprate              : 0
Interleave            : 0
Sektoren pro Spur     : 0
Sektoren insgesamt    : 16777216
Bad Sector List ab    : $000000
Lnge der BSL         : $000000
Rootsektorversion     : 104
Prfsumme             : $1234
Rootsektor von Target 0, Gert 0
    ID   Start  Size  Flag
00      000000  000000  00  
```

```
Log.Sec 000000 Cluster 000002 Track    Sektor    Side   Drive C Status -7
000000  60 5E 00 00 00 00 00 00 00 00 00 00 00 00 00 00   `^..............
000016  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
000032  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
000048  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
000064  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
000080  00 00 00 00 00 00 00 00 00 00 0F BD 01 0F B8 07   ...........œ..ž.
000096  48 7A 00 D4 3F 3C 00 09 4E 41 5C 8F 3F 3C 00 12   Hz.Ô?<..NA\?<..
000112  4E 41 54 8F 4A 00 66 22 3F 3C 00 0B 4E 41 54 8F   NATJ.f"?<..NAT
000128  4A 00 67 3A 3F 3C 00 08 4E 41 54 8F B0 3C 00 53   J.g:?<..NAT°<.S
000144  66 2C 70 03 48 40 72 0C 60 2C 76 01 61 06 76 00   f,p.H@r.`,v.a.v.
000160  61 02 60 EE 3F 3C 00 02 3F 3C 00 45 4E 41 58 8F   a.`î?<..?<.ENAX
000176  3F 00 3F 03 3F 3C 00 46 4E 41 5C 8F 4E 75 20 3A   ?.?.?<.FNA\Nu :
000192  FF 9C E0 88 72 0D 2F 00 3F 3C 00 48 4E 41 5C 8F   ÿàr./.?<.HNA\
000208  4A 80 67 00 00 5C 24 40 50 F8 04 3E 2F 0B 47 F8   Jg..\$@Pø.>/.Gø
000224  86 04 43 F8 86 06 32 BC 01 90 32 BC 00 90 11 C0   .Cø.2Œ.2Œ..À
000240  86 0D E0 88 11 C0 86 0B E0 48 11 C0 86 09 26 BC   .à.À.àH.À.&Œ
000256  00 FF 00 88 10 07 80 01 48 40 26 80 08 38 00 05   .ÿ....H@&.8..
000272  FA 01 66 F8 32 BC 00 8A 30 13 51 F8 04 3E 4A 00   ú.fø2Œ.0.Qø.>J.
000288  66 0E 0C 92 41 32 53 54 66 06 26 5F 4E EA 00 08   f..A2STf.&_Nê..
000304  20 78 00 04 4E D0 0D 41 43 53 49 32 53 54 4D 20    x..NÐ.ACSI2STM 
000320  33 2E 30 31 20 62 79 20 4A 65 61 6E 2D 4D 61 74   3.01 by Jean-Mat
000336  74 68 69 65 75 20 43 6F 75 6C 6F 6E 0D 0A 47 50   thieu Coulon..GP
000352  4C 76 33 20 6C 69 63 65 6E 73 65 2E 20 53 6F 75   Lv3 license. Sou
000368  72 63 65 20 26 20 64 6F 63 20 61 74 0D 0A 20 68   rce & doc at.. h
000384  74 74 70 73 3A 2F 2F 67 69 74 68 75 62 2E 63 6F   ttps://github.co
000400  6D 2F 72 65 74 72 6F 31 36 2F 61 63 73 69 32 73   m/retro16/acsi2s
000416  74 6D 0D 0A 53 68 69 66 74 2B 53 20 66 6F 72 20   tm..Shift+S for 
000432  73 65 74 75 70 0D 0A 00 23 07 C5 6A 00 00 00 00   setup...#.Åj....
000448  00 00 01 00 00 00 20 00 00 00 00 10 00 00 00 00   ...... .........
000464  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
000480  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
000496  00 00 00 00 00 00 00 00 00 00 00 00 00 00 55 AA   ..............Uª
```
Why is the root sector not correct?
```

/* rootsector structure */
typedef struct {
  unsigned char   prg[0x150];   /* bootprogram                                                                  */
  unsigned char   version;              /* rootsector version number            */
  unsigned long   cbhd;       /* Driver magic longword        */
  struct part     extp[8];    /* extended parts SED/ICD/Supra   */
  unsigned int    hi_cc;                        /* cylinders on disk                                            */
  unsigned char   hi_dhc;                       /* number of heads                                                      */
  unsigned char   filler;                       /* unused                                                                                               */
  unsigned int    hi_rwcc;              /* reduced current start                                */
  unsigned int    hi_wpc;                       /* write precomp start                                  */
  unsigned char   hi_lz;                        /* landing zone distance                                */
  unsigned char   hi_rt;                        /* steprate                                                                                     */
  unsigned char   hi_in;                        /* interleave                                                                           */
  unsigned char   hi_spt;                       /* sectors per track                                            */
  unsigned long   hi_size;              /* total size in sectors                                */
  struct   part   p[4];       /* atari hdx<3.0 partitions     */
  unsigned long   bsl_start;    /* bad sector list start                                */
  unsigned long   bsl_count;    /* number of sectors in bsl                     */
  unsigned int    checksum;             /* fixup for $1234                                                      */
    } ROOTBLK;

```

     
     
     

[Top](http://www.verycomputer.com/10_87aafeae5e677b51_1.htm#page-body)

Steffen
Atari STFM. 3MByte, TOS1.04 & 2.06

retro16, 2022-08-28:
Hello Steffen,

The integrated driver uses "modern" MS-DOS LBA partitions and FAT descriptors. The ROOTBLK structure you pasted seems to be for TOS format partitions. These structures aren't compatible with the MS-DOS format so your tools might be confused (I don't know SED).
Structures used by the internal driver can be found in asm/a2stdrv/structs.i

TOS does not require specific on-disk structures, all it uses is the GetBPB system function to return correct pointers to FAT12/FAT16 structures on-disk, but the header can have any format as long as the boot sector has the correct checksum. I chose to use modern MS-DOS structures to maximize compatibility with modern OSes and tools.

I'm not sure whether this answer addresses your issue, feel free to ask more precise questions if you have problems or if you want more information.

JM

Dunkelwind, 2022-09-04:
Hello JM,
tnx for the infos & clarification.
I found out that it is better to use the HATARI image for the HD emulation. This provides the correct BPB.
All the hardware-related software now works (SED, Turboass, Bugaboo....)
I think, deliver only the FAT-pointer is not good enough for a good emulation. 
Maybe you could calculate the right BPB on the fly?
But anyway, I'm using the HATARI-image now. Unfortunately, the data exchange is then more complicated
(using HATARI or loop devices)

Steffen

retro16, 2022-09-05:
The BPB is calculated on the fly: all hard disk drivers do that by hooking the GetBPB (the system call documented here: https://freemint.github.io/tos.hyp/en/bios_functions.html). In the case of the ACSI2STM internal driver, the calculation is implemented in _asm/a2stdrv/getbpb.s_.
The tools you use seem to recompute the BPB from structures on disk, but this only works for the partition types they are compatible with. If you want to check the actual BPB returned by the driver (and used by GEMDOS/TOS), you need to use the tool CHKDISK3.PRG provided by the AHDI v6 package. This tool uses the BIOS level interface to show the BPB and FAT structures as seen by the OS filesystem.

Because on-disk formats are more or less specific to their driver, you can't mix low level disk manipulation tools like you would do on PC, you have to use the disk management tool provided by the driver you use, and in the case of the ACSI2STM driver it's its setup tool you start by typing Shift+S at boot or running A2SETUP.TOS.

Now, there is a possibility that the driver actually returns a wrong BPB, and I'm interested in any potential bug feedback. For example, you mention Turboass in the things that don't work, and as far as I know this tool isn't doing low level access. Can you describe a bit more this use-case and how it fails ?

Thanks,
JM

Dunkelwind, 2022-09-09:
OK, here an example for the problem:
SD-Card:
```
Festplatte /dev/sda: 14,87 GiB, 15946743808 Bytes, 31145984 Sektoren
Festplattenmodell: STORAGE DEVICE  
Einheiten: Sektoren von 1 * 512 = 512 Bytes
Sektorgröße (logisch/physikalisch): 512 Bytes / 512 Bytes
E/A-Größe (minimal/optimal): 512 Bytes / 512 Bytes
Festplattenbezeichnungstyp: dos
Festplattenbezeichner: 0xecccb2a2

Gerät      Boot Anfang   Ende Sektoren Größe Kn Typ
/dev/sda1           32 524287   524256  256M  6 FAT16
```

show directory with GEM:
[image]

show directory with Bugaboo:
[image]

CHKDISK3.PRG:
[image]

try fopen on Bugaboo to write output to disc:

[image]

Turboass displays a src incorrectly:
[image]

Steffen

retro16, 2022-09-18:
Found the issue ! GetBPB is now fixed in the (beta) release 3.1b.
rwabs in physical mode is still known to be wrong, this will be fixed for actual stable 3.10.

Dunkelwind, 2022-09-22:
Hi JM,

I think, v3.10 is buggy too.
Even the HATARI-image stops working properly:

[image]

[image]

[image]

V3.01 with HATARI-image works fine so far  I know.

Steffen
Hint:
Manual Turboass says:
6.11 Interna des Debuggers 
`Bei Disketten-/Festplattenzugriffen: XBIOS 8/9 beim Lesen/Schreiben eines Sektors, 
XBIOS 10/18/9 beim Formatieren, 
Bei allen Dateioperationen wird (natürlich) das GEMDOS benutzt.`

retro16, 2023-06-04:
Closing as 4.00 doesn't provide a BIOS level driver anymore.

closed 2023-06-04 by retro16