retro16/acsi2stm #31: CID error... is very bad?

emilen72, 2022-09-09:
ACSI2STM SD bridge v3.01

Initializing SD0 ... CID error (36MHz) (fs ok) (rw) (25MHz 3862528 blocks) success
Searching image /acsi2stm/hd0.img ... opened
  0,0: ACSI2STM SD0 IM0   1G   

Waiting for the ACSI bus ...

Why it pass from 36Mhz th 25Mhz?

I get this error with every microSD card... can be a big problem? thanks

retro16, 2022-09-09:
Hi emilen,

The SD signal is probably a bit weak (long wire, power supply noise, ACSI bus noise, or even room lights).
If it works well at 25MHz with no read issue, you are good to go. Speed loss is minimal between 36 and 25MHz.

If you need to lock the SD card at a lower speed, you can set #define ACSI_SD_MAX_SPEED 25 (or even 16 if you want even more stable operation) in acsi2stm.h

retro16, 2023-06-04:
Hi again,
4.00 reworked a lot SD card initialization, and now uses the AdaFruit variant of SdFat, which seems to provide better SD card initialization routines than Greiman's. You should probably check the newer version.
Please provide feedback if the issue is still there.

closed 2023-06-10 by retro16