retro16/acsi2stm #33: BOM Details?

VonHirsch, 2022-12-24:
Hello, I was able to generate a BOM by importing the JSON files into EasyEDA, and then generating a BOM csv, but would you happen to have any other details like actual part numbers from a supplier or voltage requirements for the capacitors?  I'm trying to work those things out now, and I thought I'd check if you had something instead of re-inventing the wheel.  Thanks!

[BOM_PCB_ACSI2STM_EasyEDA_2022-12-24.csv](https://github.com/retro16/acsi2stm/files/10298925/BOM_PCB_ACSI2STM_EasyEDA_2022-12-24.csv)

WilsonBradley, 2022-12-25:
Sorry, trying to follow while on vacation. BOM? Not sure what this is about..

VonHirsch, 2022-12-25:
Oh sure yea I can help with that.  

> The “Bill of Materials”, commonly referred to as the BOM or electrical BOM, is simply a list. In the case of PCB design, the bill of materials is a list of all the needed parts in the manufacturing process to build that specific printed circuit board.

Source:
https://resources.altium.com/p/what-is-a-bom-in-pcb-design

retro16, 2022-12-25:
Hi @VonHirsch, thanks for reaching out.

There is no "official" BOM for the PCB, but before explaining why, answers to your questions:
* Resistors have no special requirements, anything matching roughly the value will do.
* Capacitors need to handle low power 5V, so anything over 6V is okay. I use off the shelf 0805/1206 MLCC caps.
* The EasyEDA schematic uses a weird part combining the DB19 and floppy DIN pins. I did that to get the exact spacing right when routing.

That being said, the PCB is not really designed for mass-production: some parts need manual adjustment, especially connector pins. Some parts are optional, or even mutually exclusive. I wanted to do something that supports the most diverse kind of parts possible and leave some room for tinkering.

All these choices make producing a BOM a bit pointless IMHO (I may have missed the point though, I am far from being a professional in electronics). To make a true product, the PCB probably needs to be reworked a bit to remove optional bits, make final design decisions, or even integrate the STM32 chip directly on the board to avoid the (not necessary but beneficial) blue pill board modifications.

VonHirsch, 2022-12-25:
Thanks for the answers, very helpful!

Yes, agree about the general customizability of the board build depending on preferences, and the various readme's in the repository cover those areas well.

About the BOM I guess it depends?  For the well outfitted builder pulling all these parts off the shelf it's definitely overkill.  On the other end someone making something like this for the first time could definitely benefit from a list to get them started.

As for me, this isn't my first rodeo, but I also don't have many of these parts on-hand, so I need to make an order from mouser. If the first build goes well I could see building a couple more so I'll end up organizing some sort of list.

If you think it useful I could share the list back when I have it.

retro16, 2023-07-16:
The new JLCPCB assembled "Compact PCB" available in 4.2a will provide a clean BOM from LCSC with all details. It's not blue pill based though (it has its own STM32 schematic).
This PCB is still untested, so 4.20 may or may not include this new PCB depending on test results. If everything works 100%, I will close this issue.