retro16/acsi2stm #34: ACSI2STM not showing up

bartszyman, 2022-12-30:
Hi, as for the beginning I'd like to thank you for massive work on this project. I've been following it since some time until now where I've decided to build and compile a code myself. I'm ok with soldering, schematics and general electronics.  To begin I do own 1st gen of ACSI2STM which I bought a while ago from person with nick name "Masteries" and its all ok, working fine. Next I purchased and assembled myself ACSI2SD based onCY8C5468AXI-LP042 and same it is working fine. Currently I've assembled STM32 bluebill and trying to test it with your firmware. I'm starting with 2.30 but tested 2.44 and each time results are same - all 3 of mine 1040STe's they don't see the device. In debug both cards are showing and tested with 8MHz up to 30MHz - going over is a hit and miss. Debug shows correct card info including hd0.IMG
I have spent way over a month almost every day looking online for solution, I have tried different options in acsi2stm.h #define ACSI_XXX_X  however result is always the same. This time I've decided to ask for help at the source. Can you point me to right direction please :) - the PCB board is build 100 following docs here on Github and has been tested for bridges and continuity about million times :D, additionally I've tested my bluepill by cloning the one iI own from "Masteries" and yes working OK.
[image]

bartszyman, 2022-12-31:
***UPDATE*** compiled ver.2.0 and I can see "some- incorrect" info on Atari screen meaning there is communication with device

retro16, 2023-01-09:
Hi Bart,
Thanks for the feedback. It took me a while to answer because ... well ... your case is very well documented but very difficult too.

It seems you stay stuck at "Waiting for the ACSI bus". In v2.4, the condition to exit this state and start interpreting commands is the following:
 * A1 pin high
 * CS pin high
 * IRQ pin high
 * DRQ pin high
 * ACK pin high
 * This state must stay stable for more than 100ms

Version 3.01 is much better in that respect, because it uses the RESET pin.

If you manage to go beyond the "Waiting for the ACSI bus" state, can you try to run "tos/acsitest.tos" found in the release zip file on the STE with the ACSI2STM module plugged in ? You can run the tool from a floppy.

Note: Removing R10 from your STM32 PCB can be beneficial for stability. It's a pullup on PA12 and is known to make the module unstable, especially at power on or reset. Removing the resistor will make the USB port unusable. If you want to use USB for power, you can also remove R9 and R11: these resistors connect USB data lines to the STM32, removing them will completely disconnect USB data.

Another quick note: make sure your SD card module has no logic level adapter: the STM32 works at 3.3V just like the SD card so there must be a direct connection between MOSI/MISO pins and the STM32 (with pullups to 3.3V).

bartszyman, 2023-01-13:
Thanks for the answer. I have tried FW 3.01 but result is identical, I'll try removing R10 resistor as suggested, R9 and R11 are irrelevant to my build since I don't use USB power due to PIN11 5V mod.

wolf10000, 2023-02-04:
Hi, also many thanks from my side for this fantastic development. As bartszyman I also try to build my own ACSI2STM although I do own the unit from Masteries as well and that works very fine. Its just I would like to use other harddisk drivers than the one from PPutnik (Masteries unit needs that) and I would like to use more than one SD Cards on the unit.
I came a little further than bartszyman, my unit detects the SD card and talks to the Atari but shows the following message and on the serial monitor and that's it:
  

ACSI2STM SD bridge v3.10

Initializing SD0 ... (25MHz 7802880 blocks) success
  0,0: ACSI2STM SD0 F16   4G RB

Waiting for the ACSI bus ...
--- Ready to go ---
ACSI0:

 -- Quick reset --

Waiting for the ACSI bus ...
--- Ready to go ---
ACSI0:

 -- Quick reset --

Waiting for the ACSI bus ...
--- Ready to go ---
ACSI5:4 - Not for us
ACSI5:4 - Not for us
ACSI5:4 - Not for us
ACSI5:4 - Not for us
ACSI5:4 - Not for us
ACSI5:4 - Not for us
ACSI5:4 - Not for us
ACSI5:4 - Not for us
ACSI5:4 - Not for us
ACSI5:4 - Not for us
ACSI5:4 - Not for us
ACSI5:4 - Not for us
ACSI5:4 - Not for us
ACSI5:4 - Not for us
ACSI5:4 - Not for us
ACSI1:4 - Not for us
ACSI0:

 -- Quick reset --

Waiting for the ACSI bus ...
--- Ready to go ---
ACSI0:

 -- Quick reset --

Waiting for the ACSI bus ...
--- Ready to go ---
ACSI0:

 -- Quick reset --

Waiting for the ACSI bus ...
--- Ready to go ---
ACSI0:

 -- Quick reset --

Waiting for the ACSI bus ...
--- Ready to go ---
ACSI0:

I am using the Arduino IDE to program the STM32 with an UART as you suggested, I had to set the serial monitor to 115200 as with 2000000 baud only non readable output was generated.
I also set 
#define ACSI_SD_CARDS 1
for now (until I get it working), nothing else was changed in the    acsi2stm.h    file.
In the manual you are mentioning that it is possible to program the STM32 also with the command 

    stm32flash -w acsi2stm-X.XX.ino.bin COM1:

I do have an UART which supports 5V and 3,3V (I can actually use either of them, just connecting it to the right contacts on the STM32 board), I suppose the binary file can also be uploaded with the STM32CubeProgrammer ( I use that one normally), but where from do I get the acsi2stm-X.XX.ino.bin file? Or would that one have to be compiled from the sources first?
If you can give any hints I would appreciate a lot

retro16, 2023-06-04:
4.00 should solve all problems mentioned in this issue: it looks like the STM32 hardware bug triggered by newer versions.

@wolf10000, to flash a firmware, STM32CubeProgrammer will probably work (I don't use it myself, but there is no reason it doesn't work). 4.00 will be updatable from the ST itself, so it's the last time you do a painful firmware flash operation anyway.

Precompiled bin files can be found in the release section of github (acsi2stm-4.00-release.zip below the release notes).

closed 2023-06-12 by retro16