retro16/acsi2stm #36: flashing <disk change> on every opperation on second SD card.

XR4x4, 2023-01-22:
I have an issue where when I double click on a drive partition or folder on the second SD card (drive 1) the ST's desktop flashes up <Disk Change> about 5 times then it will eventually open the drive/folder.

Trying to open a program from that sd card also gives the <Disk Change> flash before doing anything.

I have an STE, latest firmware for the stm32. SD card in slot 0 works fine. (I have tried multiple sd cards in slot 1)

retro16, 2023-06-04:
Hi, sorry for the very long wait as I somewhat missed the notification (my bad).
Anyway, better late than never.
Firmware 4.00 should solve the issue (and many others).

Can you try 4.00 and close the issue if it works ? If it doesn't I'll probably ask for more information.

Thank you for your feedback and your help.

XR4x4, 2023-06-05:
Issue is not present in current version (v4.0)

closed 2023-06-05 by XR4x4