retro16/acsi2stm #42: Can't see any files on ST, but newdesk.inf is read.

TzOk83, 2023-06-04:
The last 4.0 version in which I see files from SD card in GemDOS mode in Atari Desktop is 4.0e. In 4.0f and final release the newdesk.inf is loaded, but when I open C: disk I see no files at the card.

retro16, 2023-06-04:
Wow, thanks for the super quick update !
To investigate further, I will need your TOS version, and 3 debug (not verbose) logs of 4.00:
* booting
* opening C:
* running TOSTEST.TOS (this one should be huge, but very informative. If you can't post it easily, tell me and I will provide a link to upload it)

retro16, 2023-06-04:
Addendum: I also need to know the format of your SD card (fat / fat32 / exfat, partition count, containing long file names or not, the OS you used to format it and copy files on it).

TzOk83, 2023-06-04:
Sorry but DEBUG build doesn't work at all for me. I got only, and then PC13 starts to fade in-out:
```
ACSI2STM SD bridge v4.00

SD0 36MHz 3913728 blocks rw mountable
Waiting for the ACSI bus ...
--- Ready to go ---
``` 
It doesn't boot, however.

I tried TOS 1.04, and 2.06, FAT16, and FAT32.

It works, but I see no files in the GEM. However, if I save the desktop, the newdesk.inf file on the card is updated, and the new version is correctly read during boot.

retro16, 2023-06-04:
Okay thanks for the feedback. Will have a look.
It seems that we're going for a 4.01 in very little time, thanks to your feedback.

TzOk83, 2023-06-04:
I tried to compile 4.00 without 8.3 and dot files filtering, but I always end up with non-functional binary (with PC13 behaving like in DEBUG build - fading in and out), and not fitting into 64kB Flash (yet all mine Bluepills have 128kB).

Board: Generic STM32F103C series, CPU Speed (MHz): 72Mhz (default), Variant: STM32F103CB (20k RAM, 128k Flash), Optimize: smallest (default), Upload method: Serial.
```
Sketch uses 77336 bytes (59%) of program storage space. Maximum is 131072 bytes.
Global variables use 12240 bytes (59%) of dynamic memory, leaving 8240 bytes for local variables. Maximum is 20480 bytes.
```

retro16, 2023-06-05:
I didn't test 2.06 for 4.0f/4.00, and the regression was really subtly hidden by bugs/inefficiencies of TOS 1.62, and it wasn't covered by the test suite. 4.01 should fix issues with TOS 2.06 and have much better test coverage.

It is absolutely abnormal to end up with a 77336 bytes long binary. I tried to reinstall everything from scratch following the instructions in "compiling.md"  under Windows just to check if the platform has any impact, and I had practically the same result as the official build from my Linux machine.

XR4x4, 2023-06-05:
I'm having the same Issue (or similar) STE using TOS 2.06

ICD driver SD card in second SD slot and a gemdrive Fat32 2gb in the first slot.

I can copy files  from my ICD SD to the gemdrive SD but cannot see them in the destination window. (0 bytes used in 0 items.)  Putting the SD card in my windows PC I can see the files did actually copy over ok. (this is also with the newer 4.01)

retro16, 2023-06-05:
Thanks @XR4x4 for confirming the issue. I can't reproduce this exact case on my 2.06 STE.

Can you try with an ExFAT SD card in the GemDrive slot ? I begin to suspect an issue with FAT32 long file names in SdFat, but I need confirmation (ExFAT doesn't have long filenames in the sense of the ugly FAT16/FAT32 hack to retrofit LFN).

If it doesn't work in ExFAT mode, I need a (non-verbose) debug trace taken when the destination window is being opened, to see what is the issue with GemDrive itself.

retro16, 2023-06-05:
Oh thanks, it put me on the right track. As soon as the code finds an incompatible file name, it stops scanning. The obvious workaround is to avoid using any long file names (is there a way to prevent Windows from creating its infamous "system volume information" ?)
Maybe compiling with ACSI_GEMDRIVE_HIDE_NON_8_3 set to 0 will work for 4.01 (it didn't for 4.00+TOS 2.06, but who knows ...)

A fix is probably easy, but I really want to implement date/time and unicode character set for 4.10, which may take a week or 2.

retro16, 2023-06-06:
Fixed in 4.10, feel free to close if that's the case for you.