retro16/acsi2stm #43: Compilation issues under Windows

TzOk83, 2023-06-06:
Not really an issue with the project itself, but rather with libraries and dependencies. I've spent almost a whole day trying to solve this riddle.

1. You need Roger Clark's STM32 board definition, and you really want the clone of the MASTER repository, rather than the RELEASE version.
2. Do not change the folder name with the HW support files, it has to be called 'Arduino_STM32-master'.
3. You will need an 'arm-none-eabi-gcc' compiler/toolchain, instructions say, to install Arduino SAM boards (Cortex-M3) support, but this installs an outdated gcc 4.8.3-2014q1 version. The much newer version 7-2017q4 is included with Arduino Mbed OS RP2040.
4. After you change/update board/library support you need to close Arduino IDE and purge the Temp folder.

Now your 4.01 binary will be 57500 bytes in Flash, and will use 16408 bytes (80%) of dynamic memory.

retro16, 2023-06-06:
That's a very good finding. GCC 7 is miles ahead of GCC 4.8 in terms of optimization. Will update the doc to reflect that.

retro16, 2023-06-06:
That's now the official way to compile ACSI2STM as it doesn't seem to produce any regressions over the infamous hardware STM32 timer bug, which is my biggest nightmare.

Anyway, 4.10 solved every known problem, I played with it for a day with all the cases I had in constructive feedback, and it's rock solid.

closed 2023-06-06 by TzOk83

TzOk83, 2023-06-06:
Actually, you can go even further and download the official Arm GNU Toolchain V10.3-2021.10 and put it into a folder 'C:\Users\<User>\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\10.3-2021.10'. It will work, just like that ;)

https://developer.arm.com/-/media/Files/downloads/gnu-rm/10.3-2021.10/gcc-arm-none-eabi-10.3-2021.10-win32.zip