retro16/acsi2stm #45: Empty second SD slot issue

XR4x4, 2023-06-08:
I'm having an issue where I compile for two SD cards but I am only using one slot.
 (ste 2.06 4mb fat32 SD card)

Most of the time all I get is:

`ACSI2STM SD bridge v4.10

SD0 36MHz 31291392 blocks rw mountable
SD1 no SD
 SD1 no SD Waiting for the ACSI bus ...
--- Ready to go ---`

On the rare occasion it does boot the drive but it takes a long time especially when it gets down to "SD1 no SD" three times.

`no SD 

SD1 no SD Waiting for the ACSI bus ...

--- Ready to go ---

Stack canary 1 died. Reviving.

Stack canary 2 died. Reviving.

Stack canary 4 died. Reviving.

Stack canary 8 died. Reviving.

Stack canary 16 died. Reviving.

Stack canary 32 died. Reviving.

Stack canary 64 died. Reviving.

Stack canary 128 died. Reviving.

GDRV0:8 GemDrive boot TOS 206

SD0 36MHz 31291392 blocks rw mountable -> C: ACSI2STM SD0 F32  15G  

SD1 no SD SD1 no SD SD1 no SD -> D: ACSI2STM SD1 NO SD CARD 

Boot on C: (SD0) forward 

GDRV1:8 Ignore non-boot

ACSI2:8 Not for us

ACSI3:8 Not for us

ACSI4:8 Not for us

ACSI5:8 Not for us

ACSI6:8 Not for us

ACSI7:8 Not for us

GDRV0:E GEMDOS Pexec((14 bytes): 0 5 0 E0 9 5D 0 E0 9 5D 0 E0 9 5D): forward ........`

**I know I could compile for one just card or keep two SD's in the slot, just wondering if it was possible to ignore everything SD1 the first time it is detected as empty.**

retro16, 2023-06-08:
Hi, did you use the AdaFruit derivative of SdFat ? It's much quicker at initialization.

XR4x4, 2023-06-08:
Just the normal windows format but I don't think this is the issue here. If I have the two SD cards inserted (gemdos or ICD etc) everything is perfect and fast!, it just doesn't like it if I try to use the acsi2stm without a SD card in SD1 too.

Maybe I should just stick to one SD card then compile asci2stm for one SD or compile it for two cards and keep two cards in at all times.

XR4x4, 2023-06-09:
Ahh, yes I was using Bill Greiman's standard release 2.2.2, I have switched to bill's Adafruit Fork and it's working great. (even with an empty SD1 !!)

I'll close this now. Thank you

closed 2023-06-09 by XR4x4