retro16/acsi2stm #46: No SD Card

ChrisWIP, 2023-06-19:
Ok, the documentation doesn't really have anything on this error at all, to my surprise, but why would 'no sd card' show up?

I have an SD card installed of course, and I've tried three seperate ones. I built with blue pill and the PCB you published. 100k resistor is installed and all caps are surface mount ceramics. I haven't populated any of the floppy part of the board (don't need it).

The self tests for the board all pass fine from the ctrl+s module. I'm running v3.

[image]

Any help on this would be appreciated!

retro16, 2023-06-19:
Hi Chris, thanks for passing by.

Your issue is rather unusual, as there is not much happening during SD card initialization. It's a simple call to the SdFat library, that just does its thing with the SD card pins, routed straight into SD cards.
As I can see on your picture, you have quite a standard setup that looks good.

Things that work 100% because you already passed self tests:
* The power supply is good
* Connectivity with the ST (DMA chip, hard disk port, ...) is good

To debug this issue, I would do the following (sorry if I'm stating the obvious, you may have done this already):
* Check routing with a multimeter in continuity tester between each pin of the 1st SD slot and their matching SD card pin (see the table in [Connecting the SD cards](https://github.com/retro16/acsi2stm/blob/stable/doc/hardware.md)
* Check for shorts between all combinations of the SD card pins (you can probe on the STM32 side, it's easier)

Also, it would help a lot if you could switch to the official 4.10 debug firmware and provide an output of the serial port. 4.10 changed the SdFat library for Adafruit's variant because it has a different initialization routine, and the debug output may provide clues about what's wrong.

Side note: I had a series of defective SD card ports, hopefully that's not the issue because these things are difficult to unsolder / replace. It's not the most probable cause but I have to let you know anyway.

ChrisWIP, 2023-06-19:
Hi!

Thanks for taking the time to help!

All connections are good, they buzz out fine. No shorts. I put in an SD card extender and buzzed that out on the lengthened end, so I think the slot connectors are good, too.

Installing the new firmware, ehhhh, I can try? I could never get the 3.01 firmware to build, it just compiled trash and the step by step guide wasn't really helpful. In the end I found a prebuilt that worked, is there one for 4?

I have the release file but nowhere I can see is it documented which image I should use, verbose, legacy, etc etc 🤷 Are any of those good for my BluePill?

Serial logging... I'm kinda confused by that: I'm using CUBE to program this device, can I even accept a serial logging output stream with that?

ChrisWIP, 2023-06-20:
Oh! I got 4.10 to install onto the Pill: NO SD CARD is still sadly the order of the day. I guess I'll have to get the serial logging on the go to progress really hey?

retro16, 2023-06-20:
Yes, serial output is a must have. At least "debug", but "verbose" may be required. If you have a 64k chip, strictverbose will work for you, at least to have the initialization phase.

Firmware variants are documented in doc/flashing.md. Also, I'm very surprised you can't build firmwares yourself, the documentation has been tried and verified many times by other people.

Also, if nothing works, it might be needed to test the SdFat samples just in case.

I never used CUBE, this is an Arduino project. If you do things differently from what's in the documentation, I can't guarantee that anything will work at all.

retro16, 2023-06-21:
Thinking about your issue ... another question I have: do you use full size SD cards, or microSD adapters ?