retro16/acsi2stm #49: Problem upgrading firmware to newer than 4.10

hayesey, 2023-07-19:
Originally mentioned in here: https://github.com/retro16/acsi2stm/issues/47

I built an acsi2stm using the standard PCB design from this repo and have been using it with version 4.10 with no problems. Firmware originally loaded using a USB-ttl cable.

I tried to upgrade to version 4.2b and then 4.11, in both cases to test the RTC function which didn't work with 4.10 (known bug). 

I have loaded the firmwares using HDDFLASH.TOS/HDDFLASH.BIN and using the same USB cable but I cannot get 4.2b or 4.11 to work, the Atari doesn't detect that any hard drive is connected, the gemdos load screen doesn't appear. Also I noticed that the green LED on the blue pill board doesn't light up as it does with 4.10 (and then goes off once the ST is powered on).

I can re-load version 4.10 again and the board goes back to working normally.

I am using the default image: acsi2stm-xxxx.ino.bin in all cases.

I suspect my issue is down to something with my setup or my blue pill board as others using the same board as me (I made 10 and sold the spare 9) are able to upgrade OK.

I am going to try another blue pill board I have (which I bought a couple of years ago and use with greaseweasel).

retro16, 2023-07-19:
Thanks for opening a specific issue and trying so many things already.

I can already deduce this:
* The green LED pattern is very important. If the LED doesn't light up, it means that the ACSI2STM code doesn't run on the STM32 (or at least doesn't run correctly). Turning the LED on is literally the first instruction of the firmware.
* The ST and your board aren't the culprit because the issue appears before green LED. In fact, the green LED should appear even with the Blue Pill completely disconnected (maybe it's a good thing to check).
* You seem to have a consistent behavior between HDDFLASH and USB, so it's not a HDDFLASH issue (thank god ! this thing is **hard** to debug).

As far as I know, there is no compilation change between 4.10 and later, but that still needs to be double-checked. It's not impossible that something was upgraded behind my back.

Can you check the debug firmware (and its serial output) ?

hayesey, 2023-07-19:
Thanks for the info, I will check the debug output but this could be tomorrow now as I have a busy day today.

hayesey, 2023-08-17:
I've come to the conclusion that this issue was down to the "blue pill" board I was using, I've tried another one and that seems to work OK (can load 4.11 on to it). I'll carefully check all the soldering on the original board but perhaps it is a duff board. 

I'll close this issue now.

closed 2023-08-17 by hayesey