retro16/acsi2stm #50: ACSI2STM not showing up on Atari STE

BrettRogersUK, 2023-07-19:
Hi,

I have an Atari 1040 STe, 4MB RAM, Switchable TOS1.62/2.06 and the floppy drive swap MOD (A-B, B-A) so I can boot from external floppy if need be. I've also got a gotek as my main internal drive.

I built your design a couple of days ago but cannot get my Atari to see the module at all. I am powering with the 5v line on my FTDI Serial UART device. I have been testing with the device powered from my laptop USB through the FTDI serial so I can see what's going on.  I am using the Arduino suite v2.1.0. I have installed Roger Clarke's STM32 Library (https://github.com/rogerclarkmelbourne/Arduino_STM32) and the Adafruit fork of the SD Fat Library (v2.2.1).

I have managed to flash the 4.10 firmware and I have the power light and the activity light on when the flashing has finished. When turning on the Atari, the activity light goes out and --- Ready to go --- appears in the serial monitor. I have the BOOT0 and BOOT1 jumpers set to 0 (left) and have reset the device after programming. I have tried writing an image to the sd card with raspberry pi imager. This is the image I've used (https://atari.8bitchip.info/DiskImgPP1.html) and I have also tried adding this image as a file on freshly formatted card in the acsi2stm folder on the card. Either way it doesn't find anything at all.

I have tried using the ICDPro driver from a floppy disk and using the IDCHECK.PRG on the disk and it doesn't find any devices at all connected to the ACSI port. 

I changed the ACSI_SD_MAX_SPEED setting to 8 as I was having problems at 36MHz reading my sd cards (see below). 8MHz seems to work correctly. On the serial monitor I can see it say (this was when I had the speed set to 36MHz before setting it to 8MHz):

ACSI2STM SD Bridge v4.10

SD0 36MHz CID Error 12MHz Searching image /acsi2stm/hd0.img ... not found
3964928 blocks ro mountable bootable
SD1 error error error no SD
SD2 error error error no SD
SD3 error error error no SD
SD4 error error error no SD
Waiting for the ACSI bus ...
--- Ready to go ---

The above was when I flashed a precompiled BIN file using the stm32flash program. That's as far as it goes. I don't get anything else at all.

There are no messages on the screen when I turn the Atari on. When in TOS1.62 there is just the blank white screen until it reaches the desktop. No extra icons on the desktop just the normal A and B drive and Trash basket. In TOS2.06 it goes through the memory check and shows the desktop as normal. There isn't any activity at all on the ACSI device.

I have been trying to get this working now for the last couple of days without success so now I'm turning to you to see if you can help me. See pics below of my build.

Any help you can give me at all will be appreciated. I'm sure I have done all I can now with this. I've double, triple and quadruple checked my wiring to make sure all connections are correct as per your documentation and there are no shorts that I've found when testing with a multimeter.

I've had TWO blue pills as I thought one was maybe faulty, however,  the outcome is the same with both. Pics below show the serial monitor output, the front of my board and the rear of my board.

Thank you for all your hard work in bringing this to us.  I hope you can help.

[image]
[image]
[image]

jfceklosky, 2023-07-20:
A few questions:

-  What brand SD cards are you using?  Size?
-  How are the SD cards formatted?
-  Do you know if the DMA port on the STE is working?

retro16, 2023-07-20:
Hi,
Already, I can say that your SD card adapter board is very probably designed for 5V Arduino with a logic level adapter in the way. The STM32 is 3.3V, this probably explains issues with the SD card. You should try the SdFat examples provided in the Arduino File menu to solve your SD card issues.
You didn't solder PB1 and PB3..PB5 because your SD card appears read-only and you have 4 extra slots that don't exist.
Can you post a verbose output ?
Can you test using ACSITEST.TOS in ACSI strict mode (BOOT1=1 or the strict-verbose firmware variant) ?
It will be difficult to debug the DMA port without a logic analyzer or a digital oscilloscope.

BrettRogersUK, 2023-07-20:
> A few questions:
> 
> * What brand SD cards are you using?  Size?
> * How are the SD cards formatted?
> * Do you know if the DMA port on the STE is working?
> * What are the markings on the chip on the Blue pill?  This board looks like one I had that I could not get working (fake chip).
> 
> I am using these and they do work: https://www.amazon.com/gp/product/B07VKSVM21/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1

I am using an old Transcend 2GB SD card. I've been searching around for a list of compatible micro sd cards and I did find one somewhere but now I can't find it again.

The SD card has been formatted using GParted as Fat16 and FAT32 on different tries of the card and has also been written to using raspberry pi imager to write the image of the card I downloaded from the 8bitchip website above.

This is the first time I've used the Hard Drive port on the back of the atari ever. I've never had an hard drive for this machine so it's never been used before.

The DMA IC is a known good one named as C398739-001A. Not the so called bad DMA IC. I don't have problems reading from my gotek or from my external floppy drive.

The marking on this particular Blue pill is STM32 / F103C8T6 / GQ294 1998 / CHN GQ 225

the marking of my other Blue pill board is STM32 / F103C8T6 / 991KA 93 / MYS 808 (slashes indicate new lines on the chip)

I have tried both Blue pill boards but get the same outcome with both of them.

jfceklosky, 2023-07-20:
For the SD card reader side.  These also work fine.  They are designed to work with 5v and 3.3v and the include the resistors and caps to work correctly.

https://www.amazon.com/gp/product/B08CMLG4D6/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

jfceklosky, 2023-07-20:
Yes PB0 (for SD card 1) needs to be connected to ground to enable writing

BrettRogersUK, 2023-07-20:
Have tried it with PB0 connected to GND and I've tested the SD card function as well with the SDinfo test from the sdfat examples on the arduino suite.

These are the results (See pics below)

First pic is of sd card test as you can see its reading the sd card fine.

Second pic is of test on the atari serial output with verbose enabled. V4.10 of the firmware. 

ACSITEST.PRG shows a fail as no devices are detected.
[image]
[image]
[image]

retro16, 2023-07-20:
Based on the 2nd picture and 3rd pictures you posted (where the whole board assembly can be seen), I think, maybe, you got the pin order wrong on the DB-19.

It's really difficult for me to follow using only 2 pictures with different orientations, but let's agree that pin 1 of the DB-19 connector is the on on the bottom-right corner as seen on the 2nd picture, so it translates to top-right pin on the 3rd picture (back side), and it goes to A8 while it should go to B8. Your DB-19 seems to be horizontally mirrored.

The documentation says "looking at the male connector pins", meaning looking at the side that plugs into the ST. It's always difficult to be very clear about what you mean with pinouts ... If anyone has a better wording, I'm all ears.

BrettRogersUK, 2023-07-20:
> Based on the 2nd picture and 3rd pictures you posted (where the whole board assembly can be seen), I think, maybe, you got the pin order wrong on the DB-19.
> 
> It's really difficult for me to follow using only 2 pictures with different orientations, but let's agree that pin 1 of the DB-19 connector is the on on the bottom-right corner as seen on the 2nd picture, so it translates to top-right pin on the 3rd picture (back side), and it goes to A8 while it should go to B8. Your DB-19 seems to be horizontally mirrored.
> 
> The documentation says "looking at the male connector pins", meaning looking at the side that plugs into the ST. It's always difficult to be very clear about what you mean with pinouts ... If anyone has a better wording, I'm all ears.
> 
> Edit: the more I look at it, the less I am convinced about my reply. It's really difficult to tell.

The back side of the connector (solder side) is numbered with the pin numbers. So difficult to get mixed up with the pin numbers. 

Pin 1 of the connector does go to pin B8. 
[image]
[image]

retro16, 2023-07-20:
Okay that's it, I was completely wrong :) Sorry for the bad guidance.

Anyway, now the last resort: probing with a digital oscilloscope or a logic probe.

The trace to check is the command start byte: https://github.com/retro16/acsi2stm/blob/stable/doc/protocols.md#command-start-byte-st---peripheral

I typically use the following procedure:
* Boot the ST to the desktop and power on the STM32 (order doesn't really matter). The green LED of the STM32 needs to be off, indicating that the boot process is finished, ready to accept incoming commands.
* Place probes on CS (PB7), A1 (PB6) and IRQ (PA8). The best way is to solder extra pins just like you did on PA9/PA10.
* Set the scope's trigger on the CS pin, on falling edge.
* You need to capture a few milliseconds, maybe less, set your capture time rather short (this may need some adjustments)
* Arm single capture mode
* Reset the ST
* The CS pulse appears a few seconds after the first floppy access on TOS 1.x, after memory test on TOS 2.x
* To do multiple attempts, you just have to reset the ST again

If you only have 2 channels on your scope, proceed by 1st sampling CS+A1, then do a 2nd pass with CS+IRQ.

You should see the CS pulse with A1 low, and shortly after IRQ going low.

* If the CS pulse doesn't appear, or if the A1 line stays up, your ST probably needs repair
* If the IRQ pulse doesn't appear, the issue is on the STM32 side.

BrettRogersUK, 2023-07-20:
OK so I've had it on my oscilloscope (no idea if I've done it right or not I've not had this out for over a year and can't remember how to use it 🤣🤣🤣

Anyhow, this is what I've got showing. Top one in yellow is PB7 and Bottom one in Blue is PB6.l
[image]

Again, top one (Yellow) is PB7 and Bottom one in Blue is IRQ. 
[image]

No pulse on the IRQ line with looks of it.

Thanks,

Brett

jfceklosky, 2023-07-20:
Question:   Did you double check the DB-19 pin setup?  Make sure you don't have it set up as a female side.

BrettRogersUK, 2023-07-20:
> Question: Did you double check the DB-19 pin setup? Make sure you don't have it set up as a female side.
> 
> I have gotten burned by this many times.

Yes quite sure. Followed the pin out table on here and also the connector has the pin numbers on the plastic on the solder side of the connector too so no mistaking which pin for which.

retro16, 2023-07-20:
These traces don't look very good. Voltages seem low (I assume that you used x10 probes so 200mV/div means 2V/div ?) and the A1 pulse has no matching CS pulse.
Some of these lines are shared with the floppy controller, I don't recall the best way to make sure that you are capturing a HD access. I'll do captures myself this week-end to have a fresher memory of how it's supposed to behave.

BrettRogersUK, 2023-07-20:
> These traces don't look very good. Voltages seem low (I assume that you used x10 probes so 200mV/div means 2V/div ?) and the A1 pulse has no matching CS pulse. Some of these lines are shared with [...]

Yeah, I haven't got a clue now what to do. Floppy drive access both internal and external both work as they should do. 

I have NEVER had anything on this ACSI Port before a couple days ago so I'm not sure what it should look like or even if the port has ever worked. 

Here's a pic of the mainboard corner area where floppy controller and DMA are 
[image]

retro16, 2023-07-22:
This is a capture on my ST (nothing connected to the ACSI port) following the exact procedure I provided:

[image]

Top yellow trace is CS (PB7) and bottom trace is A1 (PB6). I have no other CS pulse even on disk access, only extra A1 pulses which don't matter.

BrettRogersUK, 2023-07-23:
> This is a capture on my ST (nothing connected to the ACSI port) following the exact procedure I provided:
> 
> [image]
> 
> Top yellow trace is CS (PB7) and bottom trace is A1 (PB6). I have no other CS pulse even on disk access, only extra A1 pulses which don't matter.

So does this mean then that there is a problem with the port on my ST? Looking at the schematic by sportnik and the kicad release if the motherboard the two chips that are directly involved with the acsi port are the 74LS244 and 74LS245 IC and then a bank of 68R resistors. If either one or both these ICs have failed for whatever reason could that stop the port from working.

retro16, 2023-07-23:
I'm no expert about the ST internals (I know a thing or 2, and more about software than hardware) so I can't really help much more.
I would recommend to focus on the CS signal as this signal is present at boot even if nothing is connected to the ACSI port and should properly trigger a scope/logic analyzer. If you want to trigger CS pulses easily, you can use ACSITEST.TOS that will trigger a few pulses with a single key stroke, without rebooting the ST each time.

BrettRogersUK, 2023-07-23:
> I'm no expert about the ST internals (I know a thing or 2, and more about software than hardware) so I can't really help much more. I would recommend to focus on the CS signal as this signal is [...]

So How do I test the CS signal without anything plugged into the port? Do I need to put the scope probe on the actual pin on the port itself inside the ST?

retro16, 2023-07-23:
Personally, I just remove the Blue Pill from the ACSI2STM socket and probe the female connector. Any way will work as long as it's connected.

BrettRogersUK, 2023-07-23:
> Personally, I just remove the Blue Pill from the ACSI2STM socket and probe the female connector. Any way will work as long as it's connected.

OK will give it a try soon. I've packaged my project up and it's ready to send to my dad who has another ST that it should work on. If it works on his then it shows there is a problem with mine. If it doesn't work on his, then I don't know what the problem could be. 

On a side note, when I connected Pin 12 of the ACSI to PA15 on the blue pill it didn't really do anything. If I connect pin 12 to the pin marked R on the blue Pill, I got the ready to go signal on the serial monitor. 

What is supposed to happen with pin PA15?

retro16, 2023-07-23:
The protocol is described in doc/protocols.md and signal names are in a table in doc/hardware.md.

"ready to go" happens when the ACSI2STM senses a steady logic high on all input pins (CS,ACK,A1)

PA15 senses reset from the ST, it is active low (low = reset, high = run normally). Connecting to "R" (reset) of the STM32 is dangerous because resetting the ST can interrupt SD card operations while PA15 guarantees that everything is flushed before resetting. You can ignore the pin completely and use the "legacy" firmware, but that one is less stable.

If connecting PA15 to 12 kills the STM32, your ST is malfunctioning.

BrettRogersUK, 2023-07-23:
> The protocol is described in doc/protocols.md and signal names are in a table in doc/hardware.md.
>
> "ready to go" happens when the ACSI2STM senses a steady logic high on all input pins (CS,ACK,A1) [...]

I don't suppose you have any images of the prototype you made before going to PCB? Just for comparison purposes.

retro16, 2023-07-23:
I don't have a picture, but there isn't anything fundamentally different from yours: a bunch of wires under a proto board. I used a modified DB25 connector. It used soldered SD to microSD adapters as microSD slots, but apart from that nothing interesting to see.

BrettRogersUK, 2023-07-23:
> I don't have a picture, but there isn't anything fundamentally different from yours: a bunch of wires under a proto board. I used a modified DB25 connector. It used soldered SD to microSD adapters as microSD slots, but apart from that nothing interesting to see.

Ok, No Problem. We'll just have to wait and see what happens with my dad's ST. See if the device works on his computer or not. I should know about Thursday time, if it gets delivered in good time.

I would like to take this opportunity to thank you for all your help. It's really appreciated. 

Regards,

Brett

BrettRogersUK, 2023-07-25:
Hi, 

Just a quick question. I don't think I've seen an answer in any of the documentation or if it is there I do apologise but what micro SD cards are best to use with the ACSI2STM device? There are so many to choose from.

Thanks,

Brett

Thanks

jfceklosky, 2023-07-25:
I use GigaStone cards without any issues:

https://www.amazon.com/gp/product/B07N78MVFY/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1

retro16, 2023-07-25:
Anything compatible with the SdFat library should work (it's more a question for SdFat devs !)
I use only reputable brands (Samsung, Sandisk, Toshiba, ...). I tried an old 64MB generic card and it worked well too, it worked at lower speed but it worked nonetheless.
Don't be afraid to use modern SDXC, the software can handle it perfectly, especially in GemDrive mode.

BrettRogersUK, 2023-07-25:
I've built another one today (MK3) and i'm still getting the same thing unfortunately. The reset line is working on PA15, when I turn the machine on it says 'quick reset' on the serial monitor and then I get the 'ready to go', whenever I reset the machine it also says 'Quick Reset' and then I get a ready to go again. Its reading the SD Card as It said that it had found the image hd0.img and opened it. I tied PB0 to ground to make it RW and I get the 'Ready to Go' but that's where it stops again. 

I'm gonna wait and see what results my dad has with the one I sent him. Hopefully it will work and then it'll probably mean there is a problem with my STe. Given the floppy works both internally and externally I don't think it could be a problem with the DMA chip. I'll see what I can find out anyway. I'll report back with what happens witht eh device I sent my dad (MK2).  Thanks again!

BrettRogersUK, 2023-07-26:
Just thought I'd leave an update here. The device arrived with my dad who has tried it in his machine and........ it does exactly the same thing as mine. Doesn't get any further than a ready to go signal. When powered on the blue pill has both the red and green lights on. As soon as the atari is switched on the green light goes out, never to return again until the machine is turned off. 

So I've now got two units and two atari's with the same problem.  I would have expected at least one unit to work correctly. 

I do not know where to go from here now. Can anyone advise what to try next?

Thanks

Brett

BrettRogersUK, 2023-07-26:
> Sorry to hear this. Are you located in the US? Perhaps I can send you a board to test with.

Unfortunately not. I'm in the UK. Thanks for the offer though.

jfceklosky, 2023-07-26:
I mentioned this before, you did double check the pin outs on the db19?  Make sure they are not reversed.

retro16, 2023-07-26:
Sorry to hear that the new one doesn't work ...
Anyway, I reviewed the serial trace of your MK3.
First, things that actually work:
* Stack canary messages are perfectly normal on first boot. It's a crude mechanism to detect stack overflows. As long as you don't see a message like this after the initial power up, you're good.
* SD card initialization is working so PB0, PB1, PB3, PB4, PB5 wiring is good.
* The "quick reset" behavior is also what it should be. It demonstrates that PA15 is correctly wired.
* LEDs work the correct way (green LED turning off when the STM32 is ready to process commands). All signals are connected.
* Your debug trace looks very normal for a first power up.
Now the thing that isn't:
* You should see some activity during the boot of the ST.
* The STM32 is waiting for a CS+A1 pulse but something doesn't work and I don't know what.
Could you try enabling ACSI_VERBOSE (or use the strictverbose prebuilt firmware image) and post its output here ?

BrettRogersUK, 2023-07-26:
> Sorry to hear that the new one doesn't work ... Anyway, I reviewed the serial trace of your MK3. First, things that actually work:
>
> * Stack canary messages are perfectly normal on first boot. It's a [...]

Here's the thing. I do have verbose output enabled in the acsi2stm.h file. I have ACSI_DEBUG=1 and ACSI_VERBOSE=1 but don't get get anything other than what I already posted up. I'm at a loss as well as it doesn't work in the other STe either no, I can't be that unlucky to have two that don't work with the same problem.

This is the same in both Blue Pill boards I have. Setting Verbose output doesn't really do anything other than what you've already seen in my screenshots. I've had debug and verbose enabled since I started this.

Thanks

Brett

jfceklosky, 2023-07-26:
Can you flash with the binaries from the release on github?

BrettRogersUK, 2023-07-26:
Hi,

Can I just ask does anyone with this ACSI2STM have the so called DMA PULLUP mod on the data bus lines of the WD1772 IC? 

The mod is explained here: https://www.exxosforum.co.uk/forum/viewtopic.php?f=17&t=1417

It is part of the manadatory fixes to do on the ST. Looking at the schematic the data lines connect to the LS74245 and come out on the other side of the same IC to go to the ACSI port. 

Not sure if his has any bearing on my problem.

Regards,

Brett

BrettRogersUK, 2023-07-26:
> Can you flash with the binaries from the release on github?

Yes I can flash binaries using the stm32flash program that comes with the Roger Clarke library I downloaded.

BrettRogersUK, 2023-07-26:
> I mentioned this before, you did double check the pin outs on the db19? Make sure they are not reversed.

Yep, Checked, double checked, triple checked and all buzzed out. As mentioned before as well the pins are numbered on the reverse side (solder side) of the connector so there is no way I could have got it mixed up. I did one wire and one connection at a time crossing each one off after I had done it following the pin our diagram under the hardware.md file on here.

jfceklosky, 2023-07-27:
The flash looks a lot like what I see, but I am using the one in here:

https://github.com/rogerclarkmelbourne/Arduino_STM32

Arduino_STM32/tools/linux64/stm32flash
stm32flash Arduino_STM32_0.9

I am running out of ideas on my side.   I know there is a whole thread on bad DMA's but to have two of them really seems odd!  I have an 520 STE (upgrade to 4MB and it does NOT have that mod installed) and two Mega ST's that my devices work fine with.

I see you have a TOS switcher board installed (I have the same on my STE as well).  Did you try with 1.62 and 2.06?

Looking at your STM32 board, this looks like one that I could NOT get working.  Mine also had the round reset button on it.  Not sure if the STM32 could be the issue.

BrettRogersUK, 2023-07-27:
> The flash looks a lot like what I see, but I am using the one in here:
> 
> https://github.com/rogerclarkmelbourne/Arduino_STM32
> 
> Arduino_STM32/tools/linux64/stm32flash stm32flash Arduino_STM32_0.9
> 
> I am running out of ideas on my side. I know there is a whole thread on bad DMA's but to have two of them really seems odd! I have an 520 STE (upgrade to 4MB and it does NOT have that mod installed) and two Mega ST's that my devices work fine with.
> 
> I see you have a TOS switcher board installed (I have the same on my STE as well). Did you try with 1.62 and 2.06?
> 
> Looking at your STM32 board, this looks like one that I could NOT get working. Mine also had the round reset button on it. Not sure if the STM32 could be the issue. Here is the amazon link to the ones I could not get to work: https://www.amazon.com/gp/product/B0BXDMT74Z/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1
> 
> **The following I have tested and they work:** https://www.amazon.com/gp/product/B0B3X4BH4Q/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&th=1 https://www.amazon.com/gp/product/B07VKSVM21/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

Well I saw that the stm32flash was included in the library but I couldn't work out how to use it under linux. I just did a sudo apt install stm32flash and it installed it. 

I tried going into the tools directory and the folder it was in and using it from there but it kept saying it couldn't find the program called stm32flash.

The thing flashes OK. I've now had two of the blue pill boards and neither of them work. I'm very quickly losing the will to live trying to get this thing working. 

By all means it appears that the controller side of things does work from what I see on the serial output, it appears to be initialising the blue pill board, initialising the sd card, reading the sd card, returning the byte size of the byte size of the card etc however its the communication from the ST that's bothering me. 

To have two STes with this fault would be just bad luck. I can't see it to be honest. Something is wrong somewhere but I can't quite put my finger on it. 

In answer to your question yes I've tried with both 1.62 and 2.06 and get the same thing. There doesn't appear to be any communication at all.

I don't know if you've seen the kicad sciatica and board by Sportnik but these have bee instrumental in me looking at where the pins of the port go and as such have led me back everytime to the 74LS245 and 74LS244 ICs (U307 and U302) on my STe board. 

But then the realisation comes that my internal gotek and external floppy work without fault. I don't know if the 245 and/or the 244 ic have anythjng to do with the floppy or not. What I do know is that the data lines go from the 245 to the WD1772 and there's also an hd_enable line that goes from the 245 ic as well to another IC near it (can't remember which one). 

But again to have two STed with these faulty ICs is just shear bad luck I can't see it happening somehow. 

Which brings us full circle back to the blue pill board. I don't know if I want to spend any more money trying to get this working. It's about £17 for two of the boards from one of those you recommend and that's coming from Amazon US. Something I can t really afford at the moment.

Unless anyone else can come up with someway to test this I'm on a lonely road. 

Thanks anyway for trying to help. It is diffuse to help remotely. 

Regards,

Brett

BrettRogersUK, 2023-07-27:
> Maybe ... maybe there is a way to test the STM32, more or less: You can simulate CS+A1 using switches (or just playing with wires if you don't have any):
> 
> * Disconnect CS (PB7) and A1 (PB6) from the DB19
> * Connect to the ST and power it on (important to keep other signals at the correct level)
> * Connect CS+A1 to +3.3V during STM32 startup
> * When the STM32 is "ready to go", put A1 to GND, then briefly CS to GND: you should see something on the log. It will be meaningless and won't do anything good, but at least it will validate that the STM32's timers are functional.
> 
> There is no risk for the ST hardware itself in doing that (maybe TOS will crash but we don't care at that point), other lines won't be activated except IRQ so it's fine because IRQ is a pullup.
> 
> If it works, then it may be a logic level incompatibility. Voltages seem to be strangely high on the ACSI port so it may work on all my machines by accident, maybe a voltage divider on CS+A1 could solve your issues.

Well I will  try the above but I've about had it now, I am about to admit defeat on this. I've no idea why it isn't working on either of the STE's I have. As far as I'm concerned I've wired it up correctly. Everything checks out end to end, but for some UNKNOWN reason it fails to work as it should. 

I think every avenue has been exhausted now trying to get this working. Either I've been very unlucky and have two STE's where the ACSI port doesn't work, Two boards that have been made up following the pinout diagram to the letter and neither of them work or Two Blue Pill boards that are fake and won't work.  There are no shorts, continuity checked end to end. Continuity checked on the Atari mainboard to make sure all tracks are connected where they should be between the port and wherever they go to. 

I have tried changing the two IC's on the mainboard the 74LS245 and 74LS244 but that made no difference either.

There is no reason at all why it shouldn't work, none whatsoever but never the less it doesn't. So either I'm extremely unlucky (which wouldn't surprise me) or there is something else wrong somewhere. Regardless, there are many others that have built this project and it works straight away so maybe I am extremely unlucky.

Time to bring this to a close now. I would like to thank everyone that has tried to help get this working for me. I know it can be very difficult to remotely try to fix something without the actual device in our hand. 

I daren't fork out the money to buy a ready made one just in case the same thing happens with that that has happened with the one I built.

Thanks again folks

Brett

retro16, 2023-07-28:
Sorry to hear that. If you or your friends ever go near Paris, just send me a message and I'll give you one of my dev units lying around.

That offer applies (limited to the available stock ;) to all the fantastic people helping the project with their constructive feedback and patient attempts at making the thing work better.

JM

BrettRogersUK, 2023-07-28:
> Been following this project for sometime having got my Mega 4 out of the loft recently and pleasantly surprised to find it still works.
> 
> First question is, Brett where did you source the 19 pin D-Type? Constructing something reliable is one of my excuses for not having got further.
> 
> 2nd. I'm in the UK and thus might be able to test one of your boards Brett. I've got a real SH204 (although not been brave enough to power this up yet) so I'm pretty confident in the HD port on the back.
> 
> Regards Pete

Thanks for the offer Pete but I've taken the decision to put this on the back burner for the foreseeable future. 

I got the DB19 connectors from ebay. They are available on there but can be expensive. I got mine last year in 2022 I think. But they are there. From what I've seen people do find it better to use a readily available db25 connector and chop 6 pins off it from one side or the other. 

Thanks,

Brett

BrettRogersUK, 2023-07-28:
> Sorry to hear that. If you or your friends ever go near Paris, just send me a message and I'll give you one of my dev units lying around.
>
> That offer applies (limited to the available stock ;) to [...]

Thank you for the kind offer. Much appreciated. I'm putting this on the back burner for the foreseeable future and shall revisit again at some point.

Thanks for all your help.

Regards,

Brett

galumph, 2023-07-28:
No worry Brett, I think I must have had my search criteria wrong before because I can see the connectors on Ebay now, just a 10'er each! Guess I should try the hack a DB25 instead then. Already got several Blue Pills so that's not an issue.

jfceklosky, 2023-07-28:
Pete

For DB19's you can remove the shell on these:
    https://www.digikey.com/en/products/detail/norcomp-inc/171-025-113R001/858139
and cut it down.

Also there is a printable DB 19 source here:
    https://github.com/CayMac-Team/DB19

Joe

BrettRogersUK, 2023-07-28:
I know I said was gonna leave it for now but I thought I'd just have a play with it. Not done anythjng to it other than plug it in again and now I'm getting this on the serial monitor when I turn the machine off.

Still not getting anything during boot but have not tried that Cs + A1 thing yet. 

What does this mean? Does it mean it hasn't detected anything on the bus?
[image]

retro16, 2023-07-28:
The [x:x] output means that it tried to read a byte. That's exactly what happens when a CS+A1 pulse is detected by the STM32.
When the ST is turned off, data lines don't go down simultaneously, so this happens if the A1 line goes low first, then the CS line, then reset, all this in very little time (it's more or less random): it matches the correct command pattern by chance.
It means that the STM32 is working fine: it's not a bad clone. You don't really need to do the CS+A1 thing yourself since it answers my question.

Now the decisive factor would be a lot of probing with an oscilloscope, many combinations, specific settings, maybe playing with a potentiometer at some point. This can take a lot of time if I'm doing this myself, so doing this through messages in a github issue is ambitious to say the least.

BrettRogersUK, 2023-07-28:
> The [x:x] output means that it tried to read a byte. That's exactly what happens when a CS+A1 pulse is detected by the STM32. When the ST is turned off, data lines don't go down simultaneously, so [...]

Yes I understand. No problem. As an aside I've managed to get the headers off the blue pill board. And I've noticed that R10 and R11 are connected together on my board. R9 (220) is connected to pin A11. R10 and R11 are connected together and go to pin A12. 

Should I remove these resistors?? 
[image]

retro16, 2023-07-28:
This is discussed in doc/hardware.md
Unfortunately for you, this is unrelated to your issues as A11/A12 are used for fast (DMA) data transfers, not commands.
Removing these resistors will disconnect the USB port.

BrettRogersUK, 2023-08-07:
Just thought I'd post a little update. I haven't yet got my device working however, I purchased a device from Masteries over on the atari-forum which I believe is a based on your code with a few changes and he has decided to make it closed source. This works first time I plugged it into my STE. 

Two things to note that are different from your pinout described in the hardware.md doc. 

1. Pin 10 of the acsi goes to PA12 and pin 14 of the ACSI goes to PA8 of the blue pill on the MASTERIES device.
2. The power LED is on when the device is turned on and the activity light comes on when the atari is turned on. On the device I built both the Power LED and Activity LED are on to begin with then when the ST is turned on the activity light goes out.

I'm starting to think that I have been very unluck to get THREE very fake blue pill boards and that is the reason that it isn't working. I have had someone independently verify that my boards don't work as well on any of the ST/STE family of machines that he has and that is the conclusion that he has come to as well.

retro16, 2023-08-07:
Hi, I just had a quick overlook at your message. You mean that somebody did something closed source based on my code ? This is strictly illegal and it needs to change immediately. I chose the GPL license for a reason: spread knowledge, encourage hardware preservation, and stop silly cash cows that we see with retro stuff lately.
Can you give me pointers so I can get in touch with its author ? Thanks.

If PA12 and PA8 are swapped, it means it's based on the horrible V1.0 code base. I can say I'm not really happy by how things go, knowing that some people continue using crippled versions, sell incompatible hardware and don't contribute to the project. It makes me really sad, and a bit angry.

Anyway, 4.2 is nearly ready, and the new PCB built by JLCPCB/LCSC (along with a proper 3D printed case) should hopefully put a stop to all these crappy attempts at building sub-par devices using shady blue pill resellers (I include my current design in this criticism as well). 4.2 will be a finished and polished product, with the exception of the first time programming that unfortunately LCSC can't do themselves. The prototypes I received last week work beautifully and look awesome. It needs a second iteration though, more on this in october if all goes well.

galumph, 2023-08-07:
Following Brett's message I Googled it and found them on Tindie (although sold out) He does link back to this github for the source but whether it's unmodified I've no idea. He claims to make an SD card image available after you've bought one to boot off.

https://github.com/retro16/acsi2stm/issues/50#issuecomment-1668423158

BrettRogersUK, 2023-08-07:
> Following Brett's message I Googled it and found them on Tindie (although sold out) He does link back to this github for the source but whether it's unmodified I've no idea. He claims to make an SD card image available after you've bought one to boot off.
> 
> [#50 (comment)](https://github.com/retro16/acsi2stm/issues/50#issuecomment-1668423158)

it links back to this github.

BrettRogersUK, 2023-08-07:
OK So I still need to get my device working with the OPEN source firmware. It looks like I have unfortunately fallen victim to buying THREE fake blue pill boards that won't work no matter what I do. Two from Amazon (one bought in 2022 and one bout last month) and one from Ebay (I have received a refund for this one). 

The one from ebay I suspect maybe fake due to having the RESRT printed in silk screen printing on the board itself. The chip used is marked up as C8T6 however when using the STM32CubeProgrammer software and ST Link it comes back with a device id of 0x414 and says its a high density and has 256kb Flash memory.

The other two come back with a device id of 0x410 but they have flash memory size of 128kb. Again both labelled up as C8T6.

retro16, 2023-08-07:
The 256kb chip is fake / mislabeled and has no chance to work.
Most C8T6 are 128kb so it's not really a way to tell if the chip is good or not.
Technically, if you swap PA12 and PA8, you could probably use the Masteries version with firmware 4.11 (legacy variant). Not sure about unused SD card CS/Lock lines though.

jfceklosky, 2023-08-07:
Brett,

Thanks for all of the updates...sorry on the fakes.

I can confirm that the following work with the official firmware from Matt:

https://www.amazon.com/gp/product/B0B3X4BH4Q/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

https://www.amazon.com/gp/product/B07VKSVM21/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

BrettRogersUK, 2023-08-08:
> 

Thanks Joe. I've purchased the set of two from hiletgo they're on amazon UK but shipping from US. 

See if they will get me working with the latest firmware.

Thanks for the recommendations.

jfceklosky, 2023-08-08:
I have ordered the hiletgo's twice and both sets were fine.

BrettRogersUK, 2023-08-16:
> 

Well I'm pleased to announce that the blue pill boards I ordered from America turned up today and I loaded one of them with the bluepill diagnostics software (https://mecrisp-stellaris-folkdoc.sourceforge.io/bluepill-diags-v1.640.html) and it turns out that the board i tested passed all the tests including STM32F103C8T6 authentication.

I had no problems connecting to it with ST link and STCubeProgrammer software and it downloaded the software without problems.

Looking at the markings on the chip on these boards they appear to be almost in exactly the same format as the chips I got from RS components with the F being at the end of the first line printed STM32F instead of starting on the second line.

These are the boards I got:
HiLetgo 2pcs STM32F103C8T6 ARM STM32 Minimum System Development Board Module STM32F103C8T6 Core Learning Board for Arduino https://amzn.eu/d/8k3vSZk

jfceklosky, 2023-08-17:
Crossing my fingers for your test.
These have worked for me 100% of the time.

[15 more replies]

closed 2023-08-27 by BrettRogersUK