retro16/acsi2stm #51: I must be doing something wrong... Help!

NullArgument, 2023-07-21:
Hello,
I am building an ACSI2STM for my Mega2 ST. I have everything wired up on a breadboard. I can upload the sketch, and the serial monitor shows the SD card as bootable. However, when I connect it to the Mega2 (using a 24-pin cable to the internal DMA port) and power it up, I receive an Invalid LUN error message. My micro SD card is a SanDisk Extreme 64Gb, imaged with Petari's HD adapted games image for TOS >= 1.04 (as is in my Mega2). Do you have any advice, or know why I would receive the Invalid LUN error?

The reason you see all the extra wiring is that I had originally wired up five readers, but I believe the computer USB port doesn't output enough amps to drive more than one, so I removed them from the breadboard.

[image]

[image]

retro16, 2023-07-21:
Hi,
Based on your log output the symptom is very clear: The ST doesn't detect the IRQ signal. It may be a wiring issue but it may also be a logic level incompatibility.

NullArgument, 2023-07-21:
I'll double check my wiring again, but everything looked good. All of the pins on the DB19 have corresponding pins on the internal DMA connector. Any recommendations on specific pins/paths to test?

What do you mean by a logic level incompatibility?

retro16, 2023-07-21:
You need to check the IRQ pin (don't remember what pin it is check docs). 
Check wikipedia or something else for logic level compatibility it's difficult to explain on a small message.

NullArgument, 2023-07-21:
You were right. It was both a wiring issue AND the IRQ line. On the internal DMA port the IRQ line is next to the RW line, which is supposed to be not connected. I had accidentally wired the IRQ line to the RW pin. I swapped it to the correct pin and it came right up!

Thank you!

closed 2023-07-21 by NullArgument