retro16/acsi2stm #52: Hardware Version and Software versions confusion

KingDaveRa, 2023-08-10:
Hello,

I'm a bit confused right now. I recently built the External boards via JLCPCB, uploading the files for EasyEDA in this repo. The boards are marked 'rev 1.1'. The firmware and docs mention version 2.x, 3.x, and 4.x, but sort of allude to hardware versions as well.

What I've built isn't working; it powers up and I get a green light on the STM32, but when I power on my 520STe the green light goes out and it's never detected. I've yet to put the serial debugger on it to find out what's going on. But I've also noticed when I disconnect the excternal 5v PSU, if I power on the STe, the ACSI2STM seems to still draw power, but no green LED. Something is very wonky, and I'm not sure if I need to make some mods to it, or if there's something else going on.Just want to make sure I've got my ducks in a row before I start off down the wrong route fixing it!

Thanks in anticipation.

XR4x4, 2023-08-10:
I have the same boards from JLC, Rev 1.1, using firmware 4.11 and two FAT32 formatted sd cards on an STE and all is working well. As long as your bluepills' are good (and is sounds like it's working ok as the green LED goes off) , i'd be enabling serial debug to see what that says.

retro16, 2023-08-11:
Hello @KingDaveRa,
Okay so let me explains how it works. It's a bit messy because of how the project evolved.

There was firmware v1.0 that didn't really work.
Then came firmware v2.x that required a hardware change (rewire 2 pins).
Then 3.x introduced the (optional) reset line. 4.x kept the same hardware as 3.x.

So in terms of hardware compatibility, there are basically 3 versions (1, 2 and 3). Hardware compatibility is now final so it's guaranteed that all future firmware revisions will work on boards that worked on 3.x.

Then comes the PCB. It is compatible with firmware 3.x and up (so it uses the final hardware compatibility). It has its own numbering system (1.0 that never came out officially, then 1.1). There will be the "compact" PCB in the upcoming 4.2 (please don't order it right now it's not fully tested yet and probably doesn't work), same stuff there will be a 1.0, 1.1, ... matching firmwares 3.x and up.

Basically, I make sure that everything is compatible within a single ACSI2STM release package (except for the bleeding edge master branch which isn't expected to work, of course).

It seems that some people have the same strange issue, what you have looks like issue #50 which has not been resolved. I'm still not sure if it's really a bad batch of blue pills or some borderline out-of-spec signal levels that work more or less by chance on most hardware and I'm just "lucky" that it works on all my computers.

KingDaveRa, 2023-08-11:
@retro16 thank you for the explanation, I was worried for a while I'd picked up an old version of the board - I've done it before with other projects!

I did find #50 on my travels and it looked similar, but I was leaning more towards a board issue, but as that seems to be ruled out now - other than me doing somehting really dumb in assembly and my inspection shows nothing - I think it might be related. My STM32s came from eBay, and they've been good in the past. I suppose maybe there are some bad ones out there. 

The only thing I've yet to figure out is how to get a serial monitor going on it. I stuck my USB-TTL into it, but the TX/RX pins match, which seems wrong to me, I'd asume they'd swap (TX-RX/RX-TX)? I then had no idea what to do beyond opening PuTTY and monitoring the port, then I didn't know what speed to use! So yeah, at a bit of a loss on that as I've never fully debugged one of these.

XR4x4, 2023-08-11:
If you are using the same USB-TTL adapter as retro16 then leave it set at 200000 but I have a different adapter that I used to use and that one refuses to communicate on anything over 115000.  (make sure it's also enabled correctly in asci2stm.h too)

BrettRogersUK, 2023-08-11:
Hi,

I ordered 3 Blue Pill boards (2 x from Amazon and 1 x from Ebay) and they all don't seem to be working for me. I did purchase a ready made ACSI device (known working) so that I could rule out a problem with my ACSI port in my STE and I sent one of my prototype boards that I built from this project to my dad to test (as you'll probably have already seen in my Issue#50). 

I think I've jus got very unlucky to get THREE that just are either fake or theres a defective chip on them. 

I did order these on recommendation but they're coming from the US so not been able to test them yet. https://www.amazon.co.uk/dp/B07VKSVM21?psc=1&ref=ppx_yo2ov_dt_b_product_details

I did exactly as you've done here, checked all my connections maybe two or three times. Continunity checked all over using Multimeter and everything checks out. There isn't any issue with my Atari Hardware as the ready made one works flawlessly. 

Hope you get it sorted soon!