retro16/acsi2stm #54: ICD can't partition ACSI drives with firmware v4.00 and up

retro16, 2023-08-25:
ICDFMT.PRG has its "Partition" button disabled. Works good with firmware 3.01.
Changes to MODE SENSE are highly suspected to be the cause.

jfceklosky, 2023-08-27:
Testing with 4.12 firmware, an acsi setup with acsi2stm/hd0.img, and ICDFMT.prg (ICD 6.55a)
ICDFMT is not able to format and returns $20 error.

I ran the same test with the HDdriver 11.  It was also not able to format the drive, but was able to partition the image into multiple drives without any issues. I was able to install the HDdriver and booting with no issues.  File copies across all of the drives were fine.

The image used for hd0.img was 2GB in size.

retro16, 2023-08-27:
Formatting is not supported as it's meaningless for an SD card (what these tools call "format" is low level format, which isn't possible even on mechanical drives these days). Maybe I could approximate it by clearing a few sectors of the card to wipe the partition table and the backup GPT if present.
After a very long night, I found the issue with ICD: a conflict in GemDrive hooks disabled the "Partition" button. ACSI strict mode solves the issue.

jfceklosky, 2023-08-27:
Makes perfect sense on the formatting side of things.  

Thanks for the explanation and the long debugging with the GEMDOS setup.

I will repeat the testiing above with an hd0.img created image with the ICD software to see that ICDFMT.prg  does allow the partitioning to work (ACSI mode).  It probably will be fine since the HDdriver software was.

jfceklosky, 2023-08-27:
I repeated the test above...partitioning does work with the ICD software and an hd0.img file, but I had to switch my device into ACSI mode only.  This confirms the debug you tracked down in the GEMDOS steup.

With ACSI + GEMDOS modes enabled, the ICD software had the partition button disabled.
With ACSI only partitioning  was fine (no button disabled).