retro16/acsi2stm #56: Some boards don't boot with V4.12.  Working fine with V4.10

mamejay, 2023-09-13:
Hi Again.
I noticed that you had V4.12 out so I decided to try it out.  I even used the HDDFLASH.TOS to update from the ST
3 of my boards that run 4.10 fine dont start on 4.12.
Anyone else had that problem?  I had to use my  STM flasher to put 4.10 back on again.
I think these were some boards I could never get working and then I tried updating them from V3 to V4 and they all started working.  The dev mentioned that he made some changes to timing which may have got these working again.

Just wondering what has changed from 4.10 to 4.12 that would cause this.

retro16, 2023-09-22:
That's a bug. I'm not sure why this happens as there are nearly zero changes in 4.12.
I had boot issues during 5.0 development, maybe I have a solution.

mamejay, 2023-09-22:
Excellent!  Yes it is strange that after 4.10 my clone boards stopped working.  If there is anything you need me to test let me know as I have boards ready.