retro16/acsi2stm #57: V5.0A Compact PCB pin header?

mamejay, 2023-09-15:
Hi Team,
I am loving the new design V5.0A compact PCB.  I want to install male pins so I can click the unit direct to the back of the ST.
I am just a little confused about which side the male pins connect to.  Is it the parts side? It looks to be the parts side to me.
If that is the case which pins can you use as I am not sure there would be enough clearance with the parts and case installed.
The bom mentioned HDR-M-2.54_1X2 which are just a standard pin header.  On testing the pins I have here are loose and it would fall out.
Thanks

jfceklosky, 2023-09-19:
You might want to consider these:

https://www.digikey.com/en/products/detail/mill-max-manufacturing-corp/3149-2-00-80-00-00-08-0/7740154

I have used these in my alt board builds

https://github.com/jfceklosky/acsi2stm-alt-hardware

mamejay, 2023-09-19:
Thanks for that.  What I am concerned is that the pins look to be installed on the parts side so would need to be pretty long top clear the parts, case and be able to connect to the DB female connector on the ST.
Do you know if the pins come out the back where there are no parts or the parts side?

jfceklosky, 2023-09-22:
Without tracing things out, I think they point out to the component side based on the image here:

[image]

mamejay, 2023-09-22:
That's what I thought.  The pins will not be long enough to reach the ST from the parts side including a case.

jfceklosky, 2023-09-22:
You can go the IDC 20 route.

I created the follow copied from the Ultra Satan cable

https://github.com/jfceklosky/idc2ribbon

retro16, 2023-09-22:
Hi, the assembled compact PCB is being tested right now (as the "Pre-release" tag indicates). It works very well without modifications.
5.0 is not final because I found a huge library bug in the firmware that makes SD card swapping unstable.

For now, I have a few observations:
* You are supposed to order the part fully assembled. If it doesn't work as assembled by JLCPCB it's a bug, if you wish to assemble it yourself (good luck with these SMD 0402 parts) then you are free to do it but you are on your own.
* J1-J9 are offset to fit in the DB19 holes. It works surprisingly well. Pins suggested by Joe won't work specifically because of this offset.
* The part is thin enough to fit on the back case of a standard STF/STE. It barely fits but it fits.
* The version in the "master" branch has slightly bigger holes for DB19 screws, don't know why the design in 5.0A had wrong holes.
* The 3D printed case needs small adjustments.

Sorry I'm really short on time these days, I really would like to finish that project once and for all, and it seems it will happen before the end of 2023.

mamejay, 2023-09-22:
> Hi, the assembled compact PCB is being tested right now (as the "Pre-release" tag indicates). It works very well without modifications. 5.0 is not final because I found a huge library bug in the [...]

So do J1-J9 pins come out of the parts side?  I would have thought you have the pins coming out of the other side to give you more clearance against the ST.

retro16, 2023-09-27:
All pins are pointing toward component side. Clearance is low, but that's a deliberate design choice:

* JLCPCB only assembles components on one side. I wanted something 100% assembled and ready to go
* It's much cleaner looking when used with no case because the back side is clean
* It's more compact. I wanted the compact PCB to be compact so it makes sense that clearance is minimized