retro16/acsi2stm #58: One of 5 boards not starting up.

cebelcher, 2023-09-18:
I had 5 compact boards made by JLCPCB - one of them gives a green LED1 with power but LED2 never comes on orange.  It is identified and flashed just fine by STM32CubeProgrammer.  Any idea where it may be getting hung up so I can complain to JLCPCB or try to fix the (tiny) component?

jfceklosky, 2023-09-19:
I am not sure that the compact board has been 100% vetted yet.

retro16, 2023-09-22:
Hi @cebelcher, thanks for your feedback and sorry for the late reply. Can you try flashing version 4.10 ?
I'm fighting against a random initialization bug in the firmware, it might be a duplicate of #56.

cebelcher, 2023-09-23:
I did try 4.10 on the one of the 5 boards that doesn't seem to boot - the behavior is definitely different - the yellow LED 2 light does blink during boot up (I'm using PPutnik's TOS 1.4), but I do not see the ASCI2STM screen that I do with the other 4 boards, and I do not get a C drive on the desktop.

Not sure if this is bad assembly or a bad component from JLCPCB.

Thanks!

retro16, 2023-09-27:
Difficult to say without doing extensive probing on the hardware (oscilloscope, logic probe, component tester). Anyway, if the behavior is different, something is wrong with the board. Possible issues:
* Bad component
* Bad solder joint
* Solder bridge (too much solder, shorting component leads)
* Clone STM32 chip (I really hope it's not the issue and that JLCPCB is a reliable provider)
* Random initialization bug in the firmware (I'm fighting something like that, but it seems 4.10 is working for everyone having issues)

Personally I built 7 units (including 5 early prototypes, but it's the same schematic, different layout) and they all work beautifully.

diegoparrilla, 2025-01-10:
I’ve assembled 50 boards, and surprisingly, the failure rate remains consistent at 1 in 5.

The device itself appears fine, and I can get it to work after a few minutes of “warming up.”

After investigating the root cause, I haven’t pinpointed the exact issue, but I suspect it might be related to improper initialization of the 8 MHz clock.

I’ve built some test boards to experiment with different capacitance values, and I’m also testing changes to the HSE startup timeout and adding delays to ensure the clock stabilizes. However, I’m still uncertain about the definitive solution.

retro16, 2025-01-11:
Thank you so much @diegoparrilla for your excellent feedback. I have no definitive explanation for what you see. Here are a few elements that could hopefully help you:

 * The board's schematic is basically the same as a Blue Pill (same component values) with extra decoupling capacitors because some SD cards can generate high current spikes.
 * I'm not 100% sure about the crystal oscillator. I took the best one I could find in terms of quality, while keeping package size and price reasonable.
 * I know that the 22pF capacitors around the oscillator are very sensitive parts. I took the value from the Blue Pill. IIRC, the crystal I use doesn't specify values in its datasheet.
 * HSE handling is outside my code. The first thing that happens in my code is turning on the activity LED (done by a global object constructor, well before calling `setup()`).
 * I always had issues with serial output at boot (missing characters) which correlates nicely with your clock initialization suspicions.
 * The board **has** to boot quickly, because it races the ST during boot. That's especially true for units with 5 empty SD slots. We don't have that much time, maybe 300-500ms at best, to be safely ahead of the ST.

diegoparrilla, 2025-01-11:
Thank you so much for your amazing feedback—it’s always a pleasure to collaborate with someone as insightful as you! After digging into the issue, I think I’ve identified the root cause, and it might actually be a relatively simple fix (fingers crossed!).  

It turns out the problem doesn’t seem to come from the **HSE initialization**, but rather the **LSE oscillator**. The 32 kHz crystal isn’t starting up properly, and it either times out or never stabilizes. Here are a couple of quick tests that have shown promising results:  

1. **Grounding Trick**: If the board doesn’t start, briefly connecting **C8** to **OSC32_OUT** with a ground wire (and then removing it) gets the board to boot. Strangely enough, pressing a finger on top of the crystal also works! But of course, we want a proper, long-term fix.  

2. **Increasing Startup Timeout**: Extending the LSE startup timeout has helped in my initial tests, though I haven’t fully validated it yet. Here’s the code I’m using in the `.ino` file:  

    ```cpp
    #ifdef LSE_STARTUP_TIMEOUT
    #undef LSE_STARTUP_TIMEOUT
    #endif

    #define LSE_STARTUP_TIMEOUT ((uint16_t)0x8000)  // Example: Increased timeout for LSE
    ```  

3. **Capacitor Values**: My calculations suggest that **22pF capacitors (C7 and C8)** might not be the optimal choice for this crystal. **12pF** seems closer to the right value. Why does 22pF still “work”? It’s not too far off, but it could be causing a slight offset. I’ll test 12pF capacitors as soon as I get them. Stay tuned!  

### The Curious 1-in-5 Failure  

Why does the oscillator fail in about 1 out of 5 cases? My current theory is that capacitor tolerances might be to blame. Capacitors have a +/- tolerance range, and a “worst-case” combination of extremes could make the oscillator fail to start. It’s just a hunch, but I’m hopeful further testing will confirm or disprove it!  

### A Note on Boot Times  

As you probably know, different Atari systems (ST, MegaST, STE, MegaSTE) have slightly different reset circuits, and their reset times aren’t exactly the same. From what I recall, the reset time is around **0.3 seconds**, which isn’t always enough for some peripherals (as I painfully discover developing the Multidevice and TOS Emulator). TOS 2.06 developers were clearly aware of this and cleverly added the Fuji logo during boot to buy a little extra time.  

The tricky part is that STM32 microcontrollers list an average LSE oscillator startup time of **1 to 4 seconds** in the datasheets—much longer than the Atari reset time. So, this might be another piece of the puzzle.  

Thank you again, @retro16, for all your incredible work! I’m a big fan of what you’re doing, and I’ll absolutely submit a PR once I’ve confirmed these changes and gotten everything working smoothly. Let’s keep making cool stuff for our Ataris!

diegoparrilla, 2025-01-11:
Here’s a quick update: all of my previously failed boards (10 in total) started working perfectly once I removed the capacitors **C7** and **C8**. With the standard firmware, there was no need to increase the startup timeout, and the startup time is excellent.

It seems that the stray capacitance from the PCB and pins alone is enough to make the oscillator work, which suggests that **C7** and **C8** might be significantly off from the ideal values.

I’ll continue testing once I receive the 12pF capacitors and share the results! 😊

retro16, 2025-01-11:
Thank you so much for all your work. Since there will be hardware changes, I'm also switching to 0805 components for the crazy people who want to solder components by hand (I don't judge, we are all crazy anyway, running ST's in 2025 ...)

For now, I put some 12pF caps but I'll wait for your feedback before sending a new test batch for PCB v1.1

I'm also thinking about doing a "super compact" board, with no RTC battery, only 1 or 2 SD slots, no IDC SATAN port and only the UART header for firmware programming.

All that will land in the next stable version.

jfceklosky, 2025-01-11:
I am one of the crazy people that like to solder small stuff :)

Thanks to all for your work on this project!

Biker1bob, 2025-01-12:
I also have boards I bought through JLCPCB. I am having failure to boot on several of them.. and inconsistent boot on the one I did get to boot. This is very interesting. Is there a way we might modify existing PCBs? I got them to build the boards... Was way cheaper than I could do it.

TzOk83, 2025-04-03:
Start with removing **C7** and **C8**, as suggested by @diegoparrilla ...

BrettRogersUK, 2025-04-21:
Hi, I've also got a board that doesn't want to boot. I have removed C7 and C8 from the board but it hasn't changed anything. It still won't boot up. I built my boards by hand, I used the sources from v5.0 and changed the cap and resistor sizes to 0805. I have programmed the board but only get the following out of the serial port. 

[image]

I have opened my own issue about this.  [https://github.com/retro16/acsi2stm/issues/96](url)

Thanks in Advance.

diegoparrilla, 2025-04-21:
> Hi, I've also got a board that doesn't want to boot. I have removed C7 and C8 from the board but it hasn't changed anything. It still won't boot up. I built my boards by hand, I used the sources from v5.0 and changed the cap and resistor sizes to 0805. I have programmed the board but only get the following out of the serial port. 
> 
> [image]
> 
> I have opened my own issue about this.  https://github.com/retro16/acsi2stm/issues/96
> 
> Thanks in Advance.

Your board is booting, which means the crystal circuit is working. Incorrect capacitors there would prevent it from booting entirely and cause it to halt. I’d suggest checking other areas of the PCB for your issue.

Biker1bob, 2025-04-21:
Do you use the square pins for the ACSI port? I found on some machines.. and different days mine (pcbway batch of 5) will exhibit the exact same thing. Also found my STE much more picky than the STFM.. no idea

BrettRogersUK, 2025-04-21:
> Do you use the square pins for the ACSI port? I found on some machines.. and different days mine (pcbway batch of 5) will exhibit the exact same thing. Also found my STE much more picky than the STFM.. I think it's poor connection possible issues.. as they work sometimes.

Yes I did use the square pins. I might change to the IDC20 connector as I've got a DB19 connector I can use on my DIY version.

jfceklosky, 2025-04-22:
Here is what an assembled version looks like that created and built:

[image]

[image]

jfceklosky, 2025-04-22:
Here is the repo for it:

https://github.com/jfceklosky/idc2ribbon

jfceklosky, 2025-04-22:
Here are the DB 19 and the hoods

https://github.com/CayMac-Team/DB19

https://www.printables.com/model/323670-parametricka-krytka-na-dsub-konektor

BrettRogersUK, 2025-04-22:
> Personally I would really avoid using the square pins when plugging into the ST DB19 port.
> 
> You can 3rd print DB-19's now and I found a set of pins that work well.
> 
> I will post my the Gerber I created for a DB19 to Ultra Satan cable

Thanks @jfceklosky I will definitely have a look at those. I wasn't too sure about the square pins and the round machine pin ones were too thin and wouldn't make good contact. 

Ill have a look at those pins on your github and get some. DB19 connectors are like rocking horse poo nowadays so it's nice that there is a DIY version to keep stuff like this alive.

Thanks again!

Biker1bob, 2025-04-25:
> Here are the DB 19 and the hoods
> 
> https://github.com/CayMac-Team/DB19
> 
> https://www.printables.com/model/323670-parametricka-krytka-na-dsub-konektor

So I see you soldered these in a pcb - is the pcb a STD 1mm hole? Most pins I find are either too small to fit the hole, or if I find one - its exactly 1mm and wont fit.. yours fit your pcb.. but is your pcb a std D-sub pin hole of 1mm?

jfceklosky, 2025-04-25:
Yes I soldered those pins right to my ICD board.

The pins do fit and the printed DB19's will fit, but it is tight fit

[image]

[image]

BrettRogersUK, 2025-05-12:
> Yes I soldered those pins right to my ICD board.
> 
> The pins do fit and the printed DB19's will fit, but it is tight fit
> 
> [image]
> 
> [image]

Hi, Ordered some of your boards and some pins so hopefully they will be here soon and I'll be able to test them. Quick Question though. I've resorted to using one of my old boards with the blue pill on (recommended by yourself (HiLetGo ones)). I've put v5.1 on there and it's reporting the SD Card as Read Only even though the pin PB0 is tied to ground. It says it with the LEGACY firmware too. When I put v4.10 on the card is found as RW. It doesn't seem to make any sense. I've changed the number of SD Cards to 1 as well seeing as I only have the one anyway.

Write lock is currently set to 2 in the Acsi2stm.h:
#define ACSI_SD_WRITE_LOCK 2

From The serial Monitor:
ACSI2STM 5.1b by Jean-Matthieu Coulon
GPLv3 license. Source & doc at
 https://github.com/retro16/acsi2stm

        SD0 50MHz 62333952 blocks ro id0 1 F1ED33DE id0 0 cached image /acsi2stm/hd0.img not found
id0 0 cached mountable id0 0 cached 

Waiting ACSI ...

Any idea what may be causing this at all? I'll let you know how I get on with the boards.

Regards,

Brett

jfceklosky, 2025-05-12:
I am not sure what's going on in your case.  

Both of my boards work fine with the 5.0 firmware.
Here are my two test boards:

[image]

[image]

jfceklosky, 2025-05-14:
@BrettRogersUK 

I just updated the first of my boards above to firmware 5.1b (acsi2stm-5.1b.ino.bin) using the TOS updater program and it is working just fine.  The jumpers are on and the SD is writeable and it is working as expected.

BrettRogersUK, 2025-05-14:
> [@BrettRogersUK](https://github.com/BrettRogersUK)
> 
> I just updated the first of my boards above to firmware 5.1b (acsi2stm-5.1b.ino.bin) using the TOS updater program and it is working just fine. The jumpers are on and the SD is writeable and it is working as expected.
> 
> With the jumper off the SD card is readonly

It looks like I'm probably have to dismantle my boards made with the bluepill boards and redo them from scratch. 

I've made my own issue (found here) with screenshots of the serial monitor output etc.

https://github.com/retro16/acsi2stm/issues/96

It's really weird. The Masteries version works perfectly but it's based on old firmware. I don't really see why the old one would work and the new one won't work. 

Thanks,

Brett

jfceklosky, 2025-05-14:
@BrettRogersUK 

Both of my boards work as designed, but when I made these, I was still learning how to build PCB's (still not an  expert by any stretch of the imagination).  
Both of the boards need a refreshed Gerber with copper plating on both sides.  This is on my to do list.   I also converted the boards to use the new EasyDA Pro from JLCPCB.

BrettRogersUK, 2025-05-14:
> [@BrettRogersUK](https://github.com/BrettRogersUK)
>
> Both of my boards work as designed, but when I made these, I was still learning how to build PCB's (still not an expert by any stretch of the [...]

My original bluepill versions are still prototype board versions with wires connecting everything up. I've been over them with a fine tooth comb and buzzed everything out pin for pin and it all checks out with the hardware page on here and pin outs. I've even run the SdFat examples from arduino to confirm the sd card can be read and that all works. Just when it plugs into the port on the atari I don't get anything at all on the screen and the serial output shows the sd card is there but nothing coming back.

It's very weird because those bluepill versions I made HAVE been working. 

Regards,

Brett

BrettRogersUK, 2025-05-17:
> [@BrettRogersUK](https://github.com/BrettRogersUK)
>
> Both of my boards work as designed, but when I made these, I was still learning how to build PCB's (still not an expert by any stretch of the [...]

Hi,

Just thought I'd let you know that I got your DB19 to IDC20 boards made and also some compact boards made and populated by JLCPCB. 

They arrived today. At first the JLCPCB board didn't work. The lights were on but no one was home. Upon checking the PCB I found a small sliver of solder bridging two of the address pins together on the STM32 chip. Removed this and jow I got a working Hard drive for my Atari. I actually prefer to use the IDC20 cable. 

Thanks again for all your help. Much appreciated. I'm just waiting on your boards coming now for the STM32 bluepill boards agan with the IDC20 connector. It saves plugging and unplugging the DB19 connector all the time. So when those come I can get them out of the prototyping boards I've got them in proper PCBs so I've got plenty of things I can use. If they work then they work, if they don't at least my compact versions work 😀.

Again, thanks for your help.

Kind Regards,

Brett

BrettRogersUK, 2025-05-18:
> Brett,
> 
> Glad to hear that all is working now. Bridges on the STM32 would be hard to spot and happy you found it, sad to hear that the bridge was done by JLCPCB.
> 
> I checked on my boards I did redo them with copper plating. I will be adding these to my github repo soon,
> 
> Joe

Hi Joe,

That's Great!! I shall keep your githubs in my favourites. 

Quick question, does your activity light on your devices stay on when you turn the ST off? My compact versions that do actually work don't have the activity light lit when the ST is off and I get a problem reloading when I next boot boot up. Unless I switch off the unit and reconnect power and turn the ST on again it boots back as it should detecting both my SD cards (one ACSI and one Gemdrive). Am I actually missing something? I've got the 5.00 firmware loaded.

Kindest Regards,

Brett

jfceklosky, 2025-05-18:
Here is what I see on my IDC 20 to asci2stm:

**Power to board using micro USB, ST off**

[image]

**ST on and booting:**

[image]

**ST off and Micro USB power to board:**

[image]

BrettRogersUK, 2025-05-18:
> Here is what I see on my IDC 20 to asci2stm:
> 
> **Power to board using micro USB, ST off**
> 
> [image]
> 
> **ST on and booting (green LED will flicker with access):**
> 
> [image]
> 
> **ST off and Micro USB power to board:**
> 
> [image]

Hi Joe,

Pretty much what I'm seeing with mine. First oic is initial power up, 2nd pic is when ST on and booting and third pic is when ST is off.

However after the third pic, when switching mine on again it doesn't detect a SD card in slot 2 any more. I can switch the ST off, pull the USB-C power and reconnect it, switch the ST on and I get both SD cards available again. Weird. But they do still work that's the main thing. I suppose if power is coming from the ST itself then it would be switched off anyway when the ST is off. Might have to look into USB power from the ST. 

Maybe one of those side cartridge USB power doodads I've seen. 

Thanks anyway.

Brett

BrettRogersUK, 2025-05-29:
Just a quick note. Bought these from Aliexpress. They fit really well and they also fit in place if the square header pins on the compact board also.

100 500 Pcs Male PCB Pin Header Press-Fit Tip 1.0 1.3 MM Diameter Brass Plated Gold 1u Tin 1.0 Bias Connector 9.5 14.5 15.0 mm
https://a.aliexpress.com/_Ewgm9qK

I got the 1.0 x 9.5mm brass ones. They do work a treat.

jfceklosky, 2025-05-29:
Thanks for the link.  I
 need more pins and if these work that would be great!

BrettRogersUK, 2025-05-29:
> Thanks for the link. I need more pins and if these work that would be great!

I used them on your adapter board and also tried them on the compact board where the square header pins go and they fit in there too. They fit very well in the acsi connector on the atari as well.