retro16/acsi2stm #59: BAD-DMA in STE finally confirmed and fixed.

TzOk83, 2023-10-21:
A bug in DMA controller behavior in early STE machines have been finally confirmed, described and fixed. Credits go to the Christian Zietz. You can find all the details at https://www.chzsoft.de/site/hardware/new-atari-ste-bad-dma-investigation/

An original topic, with a DMA fixer PCB is available here: https://forum.atari-home.de/index.php/topic,17461.0.html

lismarc, 2024-10-06:
The SCSI2STM device we built works correctly with STF but does not respond at all with our STE. 
To solve problem we pull-up data bus on WD1772. There is change that now we have two bombs on boot if ASCI2STM is connected. 

For sure we have "faulty" C025913-38  in STE. 

Do you think the problem seats in BAD-DMA chip? Is there chance the DMA fixer PCB solve it?

TzOk83, 2024-10-07:
STfm and early STe use the same C025913-38 DMA chip. For sure, there is a bug in the DMA cycle, but it is in both machines, the only difference between STfm and STe is that STfm has no buffers on the DMA port, so the timings between STfm and STe are slightly shifted. The DMA bug can be avoided in software drivers and should not affect hard disk detection.

Is your device using the built-in GemDrive driver or one of the 3rd party ACSI drivers?

P.S.
Try deleting DESKTOP.INF from your SD card, will this make your STe to boot?

lismarc, 2024-10-07:
I've tested my device with both modes: ACSI and GEMDRIVE. Always the same result - 2 bombs on boot. 
Same result after removing all manifests files DEKSTOP.INF form SD. 

You told the ACSI port in STF and STE differs the STE has additional buffer, is it possible the buffer IC is damaged?

TzOk83, 2024-10-08:
There is a slight chance of that. These chips are U302 (74LS245) for databus and U307 (74LS244) for control signals.