retro16/acsi2stm #64: 3D printed compact cases at JLC

MichiganSynthWorks, 2024-01-22:
Greetings, great work on this!

I know its a bit early but I figured some feedback from JLC may help this along, I just submitted the files for printing there and there are issues getting them printed at JLC:

1: each file could only support 1 part (shell) for printing
2: the overall wall thickness of your model is too thin, which will be definitely cracked and damaged. Please kindly check and modify your design to submit a new order. Thanks ! We kindly suggest increasing the wall thickness of the partial thin areas at least to 0.8mm, and the overall wall thickness at least to 2.0mm, which will be better. Thanks !

spawnerbr, 2024-07-22:
@MichiganSynthWorks where did you get the STL files from the case?

MichiganSynthWorks, 2024-07-22:
It was posted on the repo but looks like its been removed
_______________
Michigan Synth Works
5702 Honey Creek Cir
Haslett, MI 48840
248-716-0496
MichiganSynthWorks.com

methanoid, 2024-12-05:
[https://www.thingiverse.com/thing:6841627](url)

Maybe you can tell me if ordering assembled boards from JLCpcb they do the DB19 connector/pins? When I experimented with placing an order it didnt show in the render as being connected which suggests its not in the BOM/PickPlaCE files?