retro16/acsi2stm #65: Mother board revision is : CA400329à GX 211 VO 3489 TOS 1.62 With v5.0a & Gemdrive Not detected at all

p0ke, 2024-01-24:
Hello,

I have the latest PCB design and the original bluepill one, both works more or less like a charm but on one of my atari, the one I use, it is not detected at all (motherboard CA400329à GX 211 VO 3489 ).
I wait lot of time, 2 leds flash up (orange and red), and never shutoff, while atari is taking lot of time before showing up with A & B drive icon on desktop. 
I have been doing couple of hardware check on Atari :

- I tested the HDD port with multimeter : no issue
- I tried to switch with another DMA from an Atari that works : same results
- I Tried another 74LS245N (the one in front of the port) : same results
- I Tried HDD diagnostic tools but those never found the device.

I don't know what I can do next. 
It is like it does not exists at all.

I'm quite noob on this project. I don't know how to debug acsi2stm hardware. 
I have USB UART device I could plug (and I did but got nothing in monitor from the arduino board).

Any help would be much appreciated.

Thanks

p0ke, 2024-01-25:
Could someone tell me what makes turn off orange led ?

p0ke, 2024-02-11:
Have been setting up the device in debug mode.
It says "-> Waiting for the ACSI bus ..."
I will try to tweak the firmware, and search hardware issue.

Biker1bob, 2024-06-04:
This is the same issue I was having. I found it is the square pin headers we use now. The port connection is not the best. Try some isopropyl into the port.. maybe some dielectric grease (improves connection) I have had this on one or two machines where round pin devices work perfectly.

fplanque, 2024-08-29:
@p0ke I have the same issue. Were you able to solve it?

Thank you.

p0ke, 2024-09-04:
Hello François, not yet.
I have no clue except that it is probably linked to the fact it is not a true 1040 ste.
(That is only supposition)

Cheers p0ke

fplanque, 2024-09-05:
In the meantime, I followed @Biker1bob's advice and paid a lot of attention to the contacts.

That definitely solved one of my original issues. But I have many more to get this to work properly. My [ACSI2STM adventure from contact issues to driver issues](https://www.fplanque.com/dev/retro/atari-st-acsi2stm-mega-ste/) is here.

fplanque, 2024-09-05:
Which is sad because IMHO this is a more elegant solution (and cheaper) than the Ultra Satan...