retro16/acsi2stm #69: ACSI2STM with GOTEK

Perceval78980, 2024-04-07:
Hi everyone, 
Maybe the question asked many times, I have an ACSI2STM which works fine and a GOTEK wich works fine too.
If I start my atari 520ste with gotek only it's OK.
If I start with ACSI2STM only same thing.
But if I boot on ASCI the gotek is hidden and not work.
Wellington, how to install floppy disk image software in my partition hard disk?

p0ke, 2024-04-07:
Hello,
Did you try to recap your psu ?

p0ke

Perceval78980, 2024-04-08:
Hello. Goodyear news.
I replaced my microSD with autoboot  by another one empty juste format in fat32.
Next  plus acsi2stm with this new card and start Atari without USB key in the GOTEK.
Next step mount peripherals for show the C partition.
Next Step plus USB key incident the GOTEK and choose image disk.
Then copy files from A to C. Or installer software from A to C.
And for the PSU  no.
When I bought the Atari, it seems like new one inside. It was à miracle. Just à little yellowing on top case and key caps.
Thanks for your response.
If you have a tips for working autoboot  in C with Gotek I will be very happy.