retro16/acsi2stm #70: UART on 5.0

Biker1bob, 2024-05-29:
[image]
[image]

Sorry it is not clear to me - DO I want 5v? or no? I have given pics of how my jumpers are wired.  Nothing on 5v.  TX<>RX and GND and 3.3v and VCC on USB and both on 3.3 on header. 

Is this correct??

James

Biker1bob, 2024-05-30:
Pinout of the adapter:

* 5V
* VCC (linked to 3.3V or 5V by a jumper)
* 3.3V
* TXD
* RXD
* GND

What does this mean?? DO I want a jumper from VCC to 5v? I assume I hook up all the pins but the 5v? or do I hook them all up?? if so I cant put a jumper on anything.. SO this makes no sense.. Can I just get a diagram. I am using HW-597 as you can see in the picture. 

If I just plug it in - 5 wires as above in STM32cubeProg. I get Error Activating device.  OF course I do not have the flash jumper on because I am using the CH340 device.

Biker1bob, 2024-05-30:
So I wired it up without the 5v pin on both ends. put the flash jumper on. It worked.  Not sure why in the instructions you say that it does not need the flash jumper.

p0ke, 2024-05-30:
I would say that you don't need vcc.

retro16, 2024-12-08:
It's either 5V or the flash jumper, both do the same thing. But you have to ensure that 5V comes before 3.3V, which is the case with the "official" UART adapter mentioned in the documentation.
Closing this issue.
JM

closed 2024-12-08 by retro16