retro16/acsi2stm #73: MegaST4 compatibility plus tos 2.06

Biker1bob, 2024-06-04:
Just FYI, so far I have not been able to get gemdrive to work on TOS 2.06 on the above computer. It's an ICD product. Switchable tos decoder for ST computers. TOS 1.04 works fine. I get 3 or 4 bombs if I try tos 2.06

I have not tried to force ACSI mode yet.

James

fplanque, 2024-08-28:
Any luck to get ACSI2STM to work with TOS 2.06?

retro16, 2024-10-27:
TOS 2.06 is the main target for the testing procedure executed before each release. Because GemDrive hooks all OS calls, it doesn't really care about TOS version except old TOS 1.00 that leaks memory when launching PRG files.

Can you try 5.0c or later ? There were numerous bug fixes since old releases.