retro16/acsi2stm #74: Atari STE - 2 bombs

francouai, 2024-06-06:
Hi guys,
I'm using this ASCI2STM since few weeks now and it doesn't work like I wanted.

On my 1040STE 4mo RAM, it works fine during the first 15mns/20mns (when it is hot) but when I reset it, I'm getting 2 bombs during the boot'up process. If I switch OFF/ON, I have the same problem. 
To clear these 2 bombs error message, I have to wait 5mns...

I'm using this interface with External RPI PSU and I have already changed my microSD card. (I have the same problem without the SD card)
I have no Idea what DMA version I have.. need to open it to check.
For sure, it is not a ASCI2STM problem, since this card works perfectly on my 1040STF.
I also tried to another 1040STE and got exactly the same bombs issue.
Any help/fix will be nice. thanks!

[Sans titre](https://github.com/retro16/acsi2stm/assets/79473174/d826905a-3640-44be-96de-f63cfbcc0efb)

francouai, 2024-06-06:
[image]

francouai, 2024-06-11:
Fixed by using a "good" DMA..   but it is not logic to have this kind of problem by using "bad" DMA.
2x STE are not compatible with ACSI2STM card.

francouai, 2024-10-12:
I used another STE with the "Good" DMA (C398739-001A) and no more problem at all since this day.

retro16, 2024-12-08:
5.00 provides PIO mode for broken/bad/buggy DMA chips.
Closing this issue.

JM

closed 2024-12-08 by retro16