retro16/acsi2stm #75: ACSI2STM GemDrive not working with Atari Mega STE cache?

phjanderson, 2024-07-23:
have an ACSI2STM with 2 SD cards, one in native TOS format with ICD PRO, another in FAT32 with GemDrive. This worked fine on my Atari Mega ST.

It doesn't seem to work correctly when I connect it to my new Atari Mega STE however. It seems that it runs fine on 8mhz and 16mhz without cache, but it hangs accessing the drives with cache enabled. Everything seems to work fine so far if I disable GemDrive completely using the jumper, even with cache enabled. So it seems that the GemDrive driver causes the trouble. I tried both the 4.12 and 5.0a firmware, no difference.

I'm wondering: is the ACSI2STM GemDrive support incompatible with the Mega STE cache or is there perhaps something wrong with my Mega STE?

I have a C398789-001 DMA controller, which supposed to be the "good" version, if I understand it correctly.

fplanque, 2024-08-30:
@phjanderson I also have issues trying to get ACSI2STM to work with the Mega STE. I have the ACSI2STM in GemDrive mode by default.

I actually just took out my Mega STE after 30 years (thinking it's about time to backup its internal drive) but I think I forgot a lot about how to configure things... 

Can you please tell me:

- How you disabled cache?
- How you switched to 8 MHz?
- If you have an internal hard drive on the mega STE, were you able to use it concurrently with the Mega STE?

Thank you so much for your help !!

phjanderson, 2024-08-30:
@fplanque  This is an issue report, not a help forum, so I will try to keep it short and more in the context of workaround information for others experiencing this issue.

> * How you disabled cache?
> * How you switched to 8 MHz?

By default the Mega STE boots in 8mhz with cache disabled. The "official" way to control the cache is through XControl. So by changing the speed and cache in XControl (and saving it!) or by completely disabling XControl by renaming it to acx should bring you back to 8mhz after a reboot.

XBoot can also control the cache, so check your boot profile if it has cache enabled.

I wrote my own programs to control the CPU speed and cache instead:
https://github.com/phjanderson/MSTE_CC
But that won't help you much at this moment as you cannot copy them to your Atari without a working GemDrive.

> * If you have an internal hard drive on the mega STE, were you able to use it concurrently with the Mega STE?

I think I briefly tried it with both enabled, not sure. I believe the Atari HD has SCSI ID 0, so you need to change the jumper settings of the ACSI2STM to start on ID 1. The ACSI2STM documentation has information about the jumpers.

If you have more questions, I recommend you to make a posting on atari-forum.com or another Atari forum. Good luck!

phjanderson, 2024-08-30:
@retro16 the Atari Sidecartrige GemDrive implementation does work on a Mega STE with cache enabled, here is the code he uses:
https://github.com/sidecartridge/atarist-sidecart-gemdrive/blob/main/src/gemdrive.s

Search for MEGASTE_SPEED_CACHE_REG to see how and when he controls the CPU speed / cache. Perhaps this information is of help to you.

retro16, 2024-10-27:
Hi, sorry for the late reply, ACSI2STM was on hold and I have a few week-ends free to try to reach a stable 5.00.

GemDrive used self-modifying code so it was not compatible with CPU cache. 5.0c removed that so it should work now.

fdanapfel, 2025-02-03:
@retro16 Hello, I just got the MegaSTE internal version of ACSI2STM and I'm getting the same issue when using a SD Card in GemDrive mode with both the 5.00 and also the 5.1a firmware of ACSI2STM.

Are those versions of the firmware supposed to  include the fix from version 5.0c, or is anything else needed to make GemDrive mode work on the MegaSTE with cache enable when using version 5.00 or 5.1a of the ACSI2STM firmware on the internal MegaSTE ACSI2STM?

retro16, 2025-02-03:
Hi @fdanapfel,

All versions above 5.0c should work the same in that regard (no more self-modifying code). The code is patched by the STM32 before being sent to the Atari. There were many changes between 5.0c and 5.00, so you might have hit a regression somewhere.

Did you check for a potential conflict with your internal hard drive, if you have any ?
Could you provide a verbose output of 5.1a, so I could try to figure out where it fails ?

fdanapfel, 2025-02-03:
@retro16 Thanks for the quick response.

Since I replaced the built-in ACSI-to-SCSI card of the MegaSTE with the internal MegaSTE version of the ACSI2STM there is no internal hard drive anymore on the MegaSTE. There are also no other devices attached to the ACSI port.
So the only hard disk in the MegaSTE is the SD card on the internal ACSI2STM.

Sorry, don't know what you mean by "a verbose output of 5.1a"?

retro16, 2025-02-05:
Hi @fdanapfel,

Thanks for the extra details.

By verbose I mean connecting the board using the serial port adapter, then flashing the verbose variant of the firmware, and copy here the output of the serial monitor, leaving a bit of context (before it crashes, then just after it crashes). You can use Arduino's serial monitor or any serial terminal program, set at 1Mbps.

fdanapfel, 2025-02-05:
@retro16 Thanks for the clarification.

It will take me a while to get this all set up and working. Also not sure if I have a TTL-to-USB adapter that supports 1Mbps, is there a way to change the baud rate of the serial port to lower values without having to rebuild the firmware?

fdanapfel, 2025-02-06:
@retro16 OK, it looks like I was succesfully able to capture the requested log:
[acsi2stm_debug.txt.zip](https://github.com/user-attachments/files/18689147/acsi2stm_debug.txt.zip)

This is from two consecutive boots of the MegaSTE. On the first boot it got all the way to the desktop, but then when I tried to open the "C:" drive by double clicking the icon on the dektop it just hung.

After resetting the Atari (market with "<Resetting the MegaSTE" in the log) I tried to open XControl to see if I could change the cache setting, but then it hung as well.

Hope this helps to track down what is going wrong.

retro16, 2025-02-06:
Thanks a lot, that's perfect.
Analysis will take a long time and I'm quite busy right now, so don't expect a quick answer.

GemDrive technically works in your log because it already loads a lot of things, so it's not a trivial issue. Also, I'm not familiar with the MegaSTE so I don't know exactly when the cache is enabled. If anyone knows here, please let me know.

In the meantime, could you try with a FAT32 or an ExFAT SD card ? I didn't really test FAT16 with small cards as they are becoming rare nowadays. There's practically zero chance it solves anything, but that's a quick test. I don't need log output, just tell me if it makes any difference.

I hope we didn't hit yet another TOS bug ...

retro16, 2025-02-07:
I had a quick look and I found something. The MegaSTE is too fast for IRQ transfer in fast mode !

No need to do that FAT32 test I requested, it's really a communication issue between the STE and the STM32.

Bad news is: I can't really do much without oscilloscope / logic analyzer traces, and probably a few attempts. I don't really want to change this fast mode protocol because it will have performance impact on other machines, and I don't have enough room to fit a machine-specific protocol.

Will think about this in the next weeks. In the meantime, I can guarantee that ACSI mode is 100% unimpacted by this.

fdanapfel, 2025-02-07:
@retro16 Thanks a lot for taking a look into this. And good to know that that you already found something.

I just tested again anyway using a larger SD card formatted as ExFAT in GemDrive mode, and the behaviour is exactly the same as with the small FAT16 formatted SD card: as soon as XCONTROL activates the cache on the MegaSTE the whole system hangs.

So I can confirm that it has nothing to do with the format of the SD card.

No rush on getting this fixed. As ASCSI mode is fully working I can still use the ACSI2STM, it is just a little less convenient to get  stuff transferred to the Atari.

If you need any further assistance with testing or providing logs please feel free to reach out to me.