retro16/acsi2stm #76: What if internal HD (Mega STE)?

fplanque, 2024-08-16:
Please allow a newbie question: what if I already have an internal HD on my mega STE (with partition C: to H:) and I would like to connect the ACSI2STM and get an SD card to show up on J: in order to backup/transfer files?

- Do I simply change the jumpers to "ACSI id 1 to 5, GemDrive enabled" as shown in manual.md ?
- Can I assume my internal HD is ACSI id 0 ?
- Is there a risk of data loss on the internal HD if 2 devices use the same ACSI id?
- bonus: Can the ACSI2STM be used with ACSI id 1 to 5 even if there is no ACSI 0 in the system?

Thank you.

PS: It would be helpful for newbies like me (who haven't touched those machines for 25+ years) if there was an FAQ at the end of manual.md (and I would suggest including the 3/4 questions above ;)

fplanque, 2024-10-27:
Hello Jean-Matthieu,

Thank you very much for your answers!

In between I was able to answer some of my own questions (and raise some more). I documented my whole experience here: [ACSI2STM on Mega STE: backup of internal HD](https://www.fplanque.com/tech/retro/atari/atari-st-acsi2stm-mega-ste/) (May be useful for others ;)

Thank you for `GEMDRIVE.TOS`. This will could make everything easier.

Here's another newbie question though: when trying to upgrade my firmware using `HDDFLASH.TOS`: flashing.md says " it uses the Seagate SCSI standard command ". Does that mean the ACSI2STM should be be connected in STRICT/ACSI mode? Or should it be in GemDrive mode? Or maybe it doesn't matter?

PS: I do understand now some of my questions could not be answered in a way that works for everyone, but I believe some others could and are actually pretty important like "What happens if I use the same ACSI ID as my internal drive?" (DANGER). Therefore I believe you should do the FAQ anyway ;) It does not matter if it is incomplete or if it is redundant with some of the canonical documentation. It is a different way of accessing the information with a different mindset and different knowledge prerequisites. As a newbie to the project I was overwhelmed by the technical info in the .md files, much of which applies to people who build the adapter themselves. I did not know what to read and in which order ;) (If you don't have time and if you'd like, I can write a starting Q&A section based on my own experience and make a pull request.)

retro16, 2024-10-27:
I just read your article, it's fantastic ! Very good job !
I totally agree with your conclusion, I share your doubts about the SCSI implementation of ACSI2STM (it started as a copypaste of Hatari's drivers) as well as *all* ACSI drivers out there. The TOS layer over it doesn't help either with its fair share of bugs.

To answer your question about HDDFLASH, it will work with ACSI2STM in any mode (strict & GemDrive), on any of its IDs. I mention Seagate protocol so people can use other tool if they want, or use HDDFLASH on another compatible drive.

About your hard disk backup issue, I'm asking a question: would using a low level disk image tool to produce a disk image be an acceptable solution ? That way, you could use the image on ACSI2STM, as well as Hatari or any other emulator compatible with raw images. Would that fit your use case ? FYI Linux is able to mount AHDI images natively as it supports both FAT16 and Atari partition schemes. It requires obscure command-line incantations to make it work, but it works.
Also, working on a disk image would reduce stress on your precious vintage hard drive, as well as eliminating any risk of losing data in the process.

Other suggestion: did you try EmuTOS ? It has a very good ACSI driver built-in, and it works really well with GemDrive through `GEMDRIVE.TOS`.

About the FAQ, maybe the documentation is missing a part about how to handle disk drives in general on Atari, with its platform-specific concepts and quirks (How ACSI works, how drivers load, what drivers do, etc...). Some kind of combined tutorial with a Q&A, maybe ? That would make ACSI2STM the ultimate starting point for anybody trying to use their ST with anything else than the builtin floppy drive.

fplanque, 2024-10-27:
> I just read your article, it's fantastic ! Very good job !

Thank you! Very kind of you to say that :)

> I totally agree with your conclusion, I share your doubts about the SCSI implementation of ACSI2STM (it started as a copypaste of Hatari's drivers) as well as _all_ ACSI drivers out there. The TOS layer over it doesn't help either with its fair share of bugs.

Thanks. I was a bit worried I might have missed something or had some hardware issue that threw me off...

> To answer your question about HDDFLASH, it will work with ACSI2STM in any mode (strict & GemDrive), on any of its IDs. I mention Seagate protocol so people can use other tool if they want, or use HDDFLASH on another compatible drive.

Ok, will upgrade then and update the article accordingly.

> About your hard disk backup issue, I'm asking a question: would using a low level disk image tool to produce a disk image be an acceptable solution ? That way, you could use the image on ACSI2STM, [...]

It's interesting in it's own right but no longer necessary as I found a way to get all my files off the HD. Now I need to test and decide if it's better to continue to work with GemDrive or ACSI but having raw image adds additional complexity I don't need at this point. It's nice for me to be able to easily add/remove files from the SD card on macOS or Windows.

> Other suggestion: did you try EmuTOS ? It has a very good ACSI driver built-in, and it works really well with GemDrive through `GEMDRIVE.TOS`.

I didn't try it yet but it's on my TODO list ;)

> About the FAQ, maybe the documentation is missing a part about how to handle disk drives in general on Atari, with its platform-specific concepts and quirks (How ACSI works, how drivers load, what drivers do, etc...). 

You're right, many questions are not specific to ACSI2STM (but those questions arise for the first time when you try to use an external ACSI device -- in this case ACSI2STM -- and as a newbie you don't necessarily have a well enough understanding to know what is general and what is specific...)

> Some kind of combined tutorial with a Q&A, maybe ? That would make ACSI2STM the ultimate starting point for anybody trying to use their ST with anything else than the builtin floppy drive.

I'll give it a shot further down the road. I am now also trying IDE and SCSI drives on the Falcon, which makes the whole driver mess even more complex to sort out ;)

retro16, 2024-11-11:
Hi again François,

I had the opportunity to work a lot on the project lately. 5.00 is out, it looks very stable on all my hardware. It went through a lot of testing, especially in multi-device situations.

I think the simplest way to achieve your use case with 5.00 is:

* Set internal HDD to ACSI id 0
* Set ID_SHIFT on ACSI2STM to free id 0 (any position should work)
* If your hard disk driver allows booting GemDrive, you have nothing more to do
* If GemDrive doesn't boot, simply copy `GEMDRIVE.PRG` into your `AUTO` folder

Older GemDrive versions used an invalid protocol that disturbed other devices on the bus in very subtle (and pseudo-random) ways, this effect should be totally eliminated now.