retro16/acsi2stm #79: Atari 520-STFM - 1024kb - ACSI2STM Compact v1.0 2023-08-24

u038265, 2024-10-30:
Hello,
I tried to install the firmware version 5.0d successfully.
When I restarted the computer I could not read the microsd C and D and the program to read the RTC did not start automatically.
I went back to the firmware version 5.0c when I restarted the computer everything started working normally again, the program that reads the RTC started automatically and I could read the microsd C and D.
Do you have any idea what the problem could be?
Thanks for your work.

u038265, 2024-10-30:
[image]
With firmware version 5.0d I only have the screen where it finds the microsd but cannot read them.

retro16, 2024-10-30:
Hi, 5.0d is a pre-release, and to be fair I somewhat rushed it to make PIO mode available as soon as possible. So it wasn't tested as much as other releases. It works on my computer, but who knows what happens on yours.
You can stay at 5.0c currently, 5.0d doesn't offer any major improvement for your setup.
As-is, I don't have enough information to debug what's happening on your computer with 5.0d, I would need a verbose log output for that.
BTW, I don't think you need the UltraSatan RTC tool at boot to set the clock, it's handled internally by the GemDrive driver. UltraSatan RTC compatibility in intented for ACSI users. Anyway it doesn't hurt either, so you can do as you like.

As always, thank you for your feedback, it's very valuable information to know when things fail and it helps tremendously improving the product. Future version 5.00 aims at perfection, nothing less, so I need all the feedback I can have.

u038265, 2024-10-30:
Thank you so much.

u038265, 2024-10-30:
Thank you so much.
Have a nice day.

retro16, 2024-11-02:
5.0e should have fixed all issues, it was properly tested this time !

closed 2024-12-08 by retro16