retro16/acsi2stm #82: Compact PCB is not really suitable for hand soldering, 0805 parts would be way easier

TzOk83, 2024-11-23:
If resistors and capacitors could be changed to 0805 footprint it would be way easier to hand solder them. Hand soldering 0402 parts is an extreme sport ;) I'm not saying it's not doable, but for sure not a piece of cake, even using a microscope. There is plenty of space on the PCB for bigger footprint parts, so I don't understand this design choice.

retro16, 2024-11-23:
Hi, the compact PCB is not supposed to be hand soldered, but built with a stencil, paste and a reflow oven. 0402 was easier for an efficient layout and cheaper. I reckon there is enough room for 0805 and some of the decoupling capacitors are probably overkill, but I don't have motivation to change the design as EasyEDA made a big mess with parts updates so the source is practically useless as-is.

TzOk83, 2024-11-23:
The main problem with hand soldering this board is that the GND pads of the capacitors are solid with the ground plane, which makes them very difficult to tin with a tiny soldering tip. I had to use a preheater and extra flux. But, at least for me, preheater means no microscope...

Stencil is a huge price increase, especially if you want to make just 1 or 2 units. I once saw the source project for a compact PCB, but it disappeared from the repository some time ago. I can try to make a reflow on my preheater, with paste applied directly to the PCB, without a stencil. Recently I managed to solder the TQFP100 chip that way and ended up with just 2 shorted pins.

retro16, 2024-11-24:
The original EasyEDA sources for the schematic and PCB are the 2 json files in the pcb/Compact directory. That's what I used to produce all the othet files.

TzOk83, 2024-11-30:
I'll look at them, but I've never used EasyEDA before. By the way, I may report a success in soldering ACSI2STM Compact PCB.

methanoid, 2024-12-05:
So if we are recommended to order the boards assembled can you tell me if ordering assembled boards from JLCpcb they do the DB19 connector/pins? When I experimented with placing an order it didnt show in the render as being connected which suggests its not in the BOM/PickPlaCE files?  It may be a "me" problem as never ordered from JLCpcb or PCBway before

BrettRogersUK, 2024-12-07:
I've used the provided sources for EasyEDA and changed all the 0402 components to 0805 and it fits well. I had to do a re-route on all all the tracks but it did it. I've ordered 5 boards so will see if they work when I've soldered them up. For the connector pins (DB19) would it be best to use the normal square dupont style header pins or Turned pin round header pins?

retro16, 2024-12-08:
Very nice @BrettRogersUK.

About your DB19 pins question, here are my design criteria for the PCB assembly to help you take the right decision:
* All parts must be available for assembly at JLCPCB/LCSC, as it's one of the leaders for low cost PCBA so I want to keep that as a mandatory condition.
* Choose future-proof parts, as much as possible.
* Keep cost low.
* If you provide all files to the manufacturer, you should receive a 100% functional unit (no need to tweak anything). Firmware flashing using the "UART adapter inserted slanted in PCB plated holes" hack counts as a valid technique.

My opinion: of course, round pins are better for the health of the DB19 port, but square pins have the advantage of being much more common standard parts. There are also alignment issues during soldering operations, I like that square pins are tied together by groups of 2. Overall, I prefer round pins if feasible.

Since this change is about making hand-soldered units possible, we could add "hand soldering" to that list.

As soon as you tested the new design, feel free to do a pull request or share the design the way you want, I will review it and we could integrate it in the project. I can do the dirty work of regenerating gerber/screenshots/reference images myself.

TzOk83, 2024-12-08:
The round "machine type" goldpins are much thinner, and I'm afraid they won't ensure proper contact in the DB19 socket.

I have just built the 2nd "compact unit", this time without the help of a microscope, but still with a preheater/reflow plate.

methanoid, 2024-12-09:
Guys, you clearly are ALL miles better than I am with a soldering iron. But
am I right that I can just send the Gerbers & BOM to JLCpcb and the 3D
model and select them to manufacture and all I then need do is insert my SD
cards? That seems too easy for a soldering-illiterate like me!!!!

retro16, 2024-12-09:
@methanoid For the PCB, that's the idea. Sometimes some parts aren't available at JLCPCB so you need to substitute (which can be tricky). When I ordered mine, I tested the board as-is: flashed firmware using the "official" USB-UART dongle, plugged USB-C power and it worked first try.
The 3D model for the case isn't officially supported as I couldn't make it work 100% on all models, so I removed it from the official source, anyway feel free to experiment. IMHO the PCB protects itself pretty well from the environment so a case isn't 100% necessary.
Feel free to open a new issue if you have difficulties with ordering the parts or you are uncertain about the result, I have some time in the next weeks to help. Like other issues, provide screenshots, detailed info, ... you know the drill.

JM

jfceklosky, 2024-12-13:
I created a fork of this design here, that works with the awesome firmware created by the team here.

[git](https://github.com/jfceklosky/acsi2stm-alt-hardware)

retro16, 2025-01-14:
The new 5.1a (pre-release) has switched everything to 0805 components. The board is still untested (I didn't fabricate it yet) so try it only if you are ready to lose your money.

There is also a new "Super Compact" PCB with reduced functionality, but taking much less space and marginally lower cost.

BrettRogersUK, 2025-04-15:
Hi All, Quick update. I've started building up my 0805 version of the compact board. Does anyone have any recommendations for good solder paste and where to buy it from? I've just spent the last two hours fighting with hand soldering the STM32  IC. I tried drag soldering method with plenty of flux but got lots of blobs then had to use the solder wick which took too much solder off which in turn left legs unattached. Complete nightmare. So any recommendations for solder paste would be great thanks. Yes I do have a hot air station too just not the solder paste. Thanks.

TzOk83, 2025-04-15:
I was hand-soldering the old compact version. Good flux (I prefer Alpha RMA-7), rosin-core leaded solder (Sn60Pb40), and that's all the secret. If you insist on using solder paste and hot air soldering, look for leaded paste. I know only local products, like AG Termopasty Easy Print /Sn62Pb36Ag2/; rosin-core leaded solder: Cynel Unipress Sn60Pb40 SW26.

BrettRogersUK, 2025-04-15:
> I was hand-soldering the old compact version. Good flux (I prefer Alpha RMA-7), rosin-core leaded solder (Sn60Pb40), and that's all the secret. If you insist on using solder paste and hot air [...]

I've got good rosin core leaded solder 60/40 it just wasn't working for me today. I usually don't have a problem hand soldering small units like the STM32 but today I just couldn't get it right for some reason. I've been wanting to try using solder paste for a while but wasn't sure which is a good one to get. Finding leaded stuff now in the UK is nearly impossible as they won't sell leaded solder or solder paste to hobbyists like myself, they'll only sell to professionals apparently. It has certainly not taken me two hours to just do one thing like the SMT32 before. Usually takes me around 2 hours to build a whole board taking breaks etc in between components.

TzOk83, 2025-04-15:
My other hint is: don't order PCBs with a white solder mask. It is thicker than regular green or red, and makes hand soldering way harder.

BrettRogersUK, 2025-04-15:
> My other hint is: don't order PCBs with a white solder mask. It is thicker than regular green or red, and makes hand soldering way harder.

Mine are green. Maybe I'll build this one seeing how I go with it. Should be ok now the SMT32 is on. Be happy if it works 🤣🤣🤣. Only need one but have another 4 boards and one more smt32 to use along with other components so in theory I can make another if this one fails.

jfceklosky, 2025-04-16:
If you don't want to deal with the STM32 soldering you are free to use my TH designs.  They work fine with the 5.0 firmware.

https://github.com/jfceklosky/acsi2stm-alt-hardware

BrettRogersUK, 2025-04-16:
> If you don't want to deal with the STM32 soldering you are free to use my TH designs. They work fine with the 5.0 firmware.
> 
> https://github.com/jfceklosky/acsi2stm-alt-hardware

Thanks for that. I don't usually have problems doing those surface mount chips but today was an exception. I'll crack on later today and finish the board see if it works. I've got about three versions using the Blue pill boards but they're home made on breadboard. I want something more permanent. So, if this doesn't take off like I'd hoped I'll get some of your boards made and do it that way. Thanks again folks!

BrettRogersUK, 2025-04-21:
> If you don't want to deal with the STM32 soldering you are free to use my TH designs. They work fine with the 5.0 firmware.
> 
> https://github.com/jfceklosky/acsi2stm-alt-hardware

Hi,

I've finished soldering my board up and got it programmed but now I have a new issue that I just can't fathom out.

https://github.com/retro16/acsi2stm/issues/96

if anyone has any ideas for me to try out I would be grateful to see if I can get this working properly.

Many Thanks.