retro16/acsi2stm #83: Clock in mixed mode, timestamp on ACSI partition

FlathVader, 2024-11-29:
My setup:

Atari 1040 STE, TOS 1.62
ASCI2STM firmware 5.00
mixed mode, PP driver

If I boot my Atari and use US_GETCL.PRG in the AUTO folder the RTC date is correctly displayed in the XCONTROL.ACC.
If I create a folder/file on the GemDrive partition the date is correctly set. 

However, if I create a folder on an ACSI partition the date is wrong.

I can correct the date manually doing the following:
If I go into XCONTROL.ACC and set the date with the one already displayed (i.e. actually no change) and again create a file/folder on an ASCI partition the date is correctly set.

If I well understand this action in XCONTROL.ACC  should trigger the GEMDOS function 43 (Tsetdate).
If I try to copy this behavior with a small program in the AUTO folder - Tsetdate(Tgetdate()) - it does not do the job and I don't understand why.

Cheers
Peter

FlathVader, 2024-11-29:
Thanks for your reply.
I did a quick test with a floppy disk - same behavior.
I may do more tests the next days.

retro16, 2024-11-30:
I found the root cause.

If I call Tsetdate/Tsettime during GemDrive initialization (before system calls are hooked), it fixes your issue.
No idea why your little program doesn't fix the problem, it really should because GemDrive forwards Tsetdate/Tsettime to the OS.

Anyway, I patched master branch and I joined a firmware image here if you don't want to rebuild the whole code yourself. This version is marked 5.0h internally. Flashable using serial or HDDFLASH as you wish.

Fixed firmware: [acsi2stm-5.0h-fix.zip](https://github.com/user-attachments/files/17965148/acsi2stm-5.0h-fix.zip)

I tested without US_GETCL.PRG so it's really unnecessary in your case (as long as GemDrive boots, it handles everything by itself internally even if there is no SD card inserted). This program is only relevant if you want to boot in strict mode.

This was not fully tested with multiple devices, so I won't make a release as-is (I must rework test procedures a lot to take this issue into account). Feel free to use that version without worry, changes were minimal so you will keep the same stable experience 5.00 brought.

FlathVader, 2024-11-30:
Thanks, the situation is now (without US_GETCL.PRG):

1/ Booting v5.0h
2/ Create new folder on ACSI partition: timestamp 30.11.28
3/ Open Xcontrol.acc, it shows 30.11.2024
4/ Set 30.11.2024 with Xcontrol.acc
5/ Create new folder on ACSI partition: timestamp 30.11.24

Same on Floppy Disk.

I checked with Sys-Mon on my Atari and I can see that Xcontrol.acc calls XBIOS Settime in addition to the GEMDOS calls Tsetdate/Tgetdate.

FlathVader, 2024-11-30:
I tried to copy Xcontrol.acc's way to set the date with a small program, now using XBIOS Settime.
No success for the year timestamp, it remains 2028. Y2k issue. Time, day, month are ok.
I am really wondering why Xcontrol.acc successfully sets the timestamp.

```
start	
	move.w	#$2a,-(sp)		; Tgetdate
	trap	#1
	addq.l	#2,sp
	move.w	d0,d1			; time in d1

	move.w	#$2c,-(sp)		; Tgettime
	trap	#1
	addq.l	#2,sp			; date in d0

	move.w	d0,-(sp)
	move.w	d1,-(sp)
	move.w 	#$16,-(sp)		; Settime
	trap 	#14
	addq.l 	#6,sp

end	move.w  #0,-(sp)		; function # to exit program.
	trap    #1			; call the operating system.
```

retro16, 2024-11-30:
To fix the first issue on 5.00, I could reproduce it on a floppy before fixing it. However I can't reproduce the 2028 issue with 5.0h no matter how hard I try.

I have a 1040 STE, switchable TOS 1.62 FR/1.62 US/2.06 FR/EMUTOS 1.2, GemDrive 5.0h, no boot program except GemDrive.

I have correct behavior for all TOSes with 5.0h when creating folders on a floppy, date shows as 2024.

No idea where this year 2028 comes from.

retro16, 2024-11-30:
After reviewing your code, I'm not sure if d1 is preserved by the Tgettime call. It is preserved by GemDrive, not sure about TOS. C calling conventions specify D0-D2/A0-A2 as scratch registers ...
Maybe you should save it in memory to be sure ? If you swap Tgettime and Tgetdate, you could push Tgettime's return value directly on stack, it will be in correct order for Settime.

FlathVader, 2024-11-30:
I did a test without the ASCI2STM device.
Just plain TOS 1.62 and XCONTROL.ACC.
If I set the date and create a folder the timestamp is correct.

=> This has nothing to do with ASCI2STM and you can close the issue.

Still no clue why XCONTROL does the thing and my test programs not.
Will try your hints, it's been 30 years since I did coding on an Atari ST :)

FlathVader, 2024-11-30:
I found the tool that messed up my system: XBOOT.PRG in the AUTO folder.
Thanks :)

TzOk83, 2024-12-01:
Don't know if it is any helpful, but year 2028 means "date not set". There is a bug in TOS 1.06 and 1.62, with synchronizing the system clock from IKBD, and it is fixed by STEFIX.TOS in AUTO folder. This fix works for both 1.06 and 1.62, despite on the latter one, program indicates that it is not needed for that version of TOS.

retro16, 2024-12-01:
That's interesting information.

I simplified GemDrive's logic for the next coming release: It simply calls Tsetdate/Tsettime at boot to synchronize the ST with its own RTC. It doesn't hook Tgetdate/Tgettime anymore. I retained the logic that sets the internal RTC using Tgetdate/Tgettime if date is not set in the ACSI2STM, so units without backup battery work correctly on computers with an internal clock.

If I understood correctly, it means that STEFIX won't be needed with the next GemDrive because GemDrive sets the date at boot. Don't quote me on that, I'm no IKBD expert.

FlathVader, 2024-12-01:
Thanks for the hint with STEFIX, I will give it a try.

As it seems to be a plain TOS bug, the following fixed it for me:
 - In https://www.atari-forum.com/viewtopic.php?t=31625 you can find, among other useful information, the link to Y2KPATCH.ZIP.
 - I downloaded it and patched a TOS image
 - Booted with that image, all good.

Now I need to get this on 2 eproms and I will be as happy as I can be :)

cathrynm, 2025-07-13:
For me that Y2KPATCH.ZIP file worked also. Just updated roms from the patch, and nothing else in AUTO aside from US_GETCL.PRG. This was a handy use for my TOS Emulator which allowed me to get the clock working without having to mess about with EPROM programming.