retro16/acsi2stm #84: Atari TT and MegaSTE versio

math05, 2025-01-12:
Does MegaSTE version work with TT? Can I safely connect it to my TT?

retro16, 2025-01-12:
Hi Math,

I never had a TT nor a MegaSTE myself, so take my reply with a grain of salt. If anybody has actual experience with these machines, feel free to validate or correct me.

The MegaSTE PCB is supposed to replace the internal ACSI HDD of the MegaSTE. AFAIK, the TT has a SCSI HDD, it is incompatible with ACSI.

You have 2 solutions:
* Use a normal "ACSI2STM Compact" unit on the external ACSI port, same as a ST/STE.
* Use BlueSCSI to emulate an internal hard drive (never tested this, I don't know if it works, you should contact BlueSCSI people).

JM

longissor, 2025-01-12:
Hi Math05,

_**JM**, sorry to highjack your message board. Let me now if you want me to delete.
And thank you very much for your work on this system._

The TT does not have internal ACSI port like the MegaST and MegaSTE. So you cannot connect the ACSI2STM internally inside a TT.
Your solution is the external compact version. You just need to get the 2 serial ports present on the VME out of the way. You can do this using this alternative VME plate: https://www.thingiverse.com/thing:6903218

For MegaSTE, you could use the cable designed for the UltraSatan and solder a 20 pin IDC connector on the ACSI2STM compact. 
But the easiest would be to do like the TT and connect at the back after moving the serial port.
You will see a photo of an ACSI2STM at the back of a MegaSTE at the end of this page: https://www.16-32bit.eu/acsi2stm

Or get the ATCI2STM version for MegaSTE here: https://www.16-32bit.eu/acsi2stm-mega-ste
Stock is back in 2 weeks

For MegaST, the compact ACSI2STM is good. Or you can 3D print a support and mount it inside using the UltraSatan cable.
more here: https://www.16-32bit.eu/acsi2stm-megast

Note that the ACSI port in the MegaSTE is different than the one in the MegaST. 

Cheers
Cyril

retro16, 2025-01-12:
No problem @longissor, this message board is public, and providing information is always welcome, especially when it's that good !

I'm working on a new "Super compact" PCB that may fit the TT without modification: it extends only 20mm above the center of the DB19 screws. To make it smaller, it has only one SD slot and doesn't have a backup battery.

[image]

jfceklosky, 2025-01-16:
Really nice compact redesign!

math05, 2025-01-17:
Thanks everybody for comments. I have a clear picture now :)