retro16/acsi2stm #85: Multiple SD cards in a compact PCB

CarstenOtto63, 2025-01-12:
I've got 3 compact ACSI2STM PCBs latest version. One ready built from ebay, 2 via the suggested PCB service. The good news, all 3 boards are working, SD cards are recognised in every slot individually and with F16/F32/image format.

But, as soon as I insert 2 SD cards at the same time, slot 1&2 for example, either no or only one card is recognised and the other slots are reported as No card.

What am I doing wrong?

I am working on an unmodified Mega ST 4 with TOS 1.4.

Thanks for your help

diegoparrilla, 2025-01-12:
What power supply are you using?

retro16, 2025-01-12:
Hi Carsten,

This is not normal behavior, I never had such issues.

What brand are your SD cards ? They might have timeout issues.

In GemDrive mode, does hot plug work ? To test:

 * Boot with no card
 * Install icons for C:, D: and E:
 * Insert a card in C:, check that it works
 * While leaving the card in C:, insert another one in D:, check that it works

The issue you have is probably independent from TOS version or machine variant.

If you can, it would be good to flash the debug firmware and paste a log of the boot here. If you need assistance for that, just ask for more help. You will need a USB-serial dongle to make this work.

CarstenOtto63, 2025-01-12:
@diegoparrilla : I am using a standard USB power supply. Should be sufficient to what I have seen about power consumption.
@retro16: I will test that immediately, up to now I always tried to boot with the cards inserted.

CarstenOtto63, 2025-01-12:
I did the tests.
After the boot without card 
* I inserted a card in each slot -> worked
* I inserted that card in slot 1 and a second card in slot 2 and 3 -> didn't open
* I removed the card from slot 1 and the drive with the second card -> drive shows up
[acsi2stm test1.txt](https://github.com/user-attachments/files/18391268/acsi2stm.test1.txt)

retro16, 2025-01-14:
Thanks for the log. The output looks like if you removed the first card (which you didn't). That's a hardware failure, and a rare one.

Possible things that have a problem:

* USB power supply
* SD card pullup resistor
* SD card itself that doesn't strictly follow the standard
* A short circuit (soldering problem, misaligned component, bent pins) on SD slots

CarstenOtto63, 2025-01-14:
Thank you for the analysis.
* USB power supply, it is a standard USB supply and alternatively power via USB-Serial adapter, I will check the voltage on the board.
* Pullup resistor: Two boards came from jlcpcb, I had to hand solder the SD slots, one from cbmretro ready-to-use. I will try to check the signals, but all boards show the same behaviour.
* The SD cards are from intenso, 4GB cards, they are SDHC cards. I thought they should be ok, but I will try with a different brand as this would be the most likely reason.
* short circuit, could be, that is why I ordered the ready-built PCB as a reference. 

It is not essential for me to have that feature, but it would be nice to have or at least know why it is not working.

CarstenOtto63, 2025-01-14:
Thanks again for your support, 
I tested it with 32GB SanDisk Extreme cards and they worked as they are supposed to. Hotswap ok, 2 cards in any slot combination works ok. The discs I used before immediately crash the system. I haven't expected that, as the SD cards are brand new.